Hexxeh/rpi-update #157: Return codes in case of update or not...

mbocquet, 2015-01-04:
At this time, rpi-update returns 0 if a new firmware has been successfully installed... and also 0 if no firmware update is needed.
Returning a different code should be better to check if we need to reboot or not in case rpi-update is launched by cron.

For example, with the current version, the following command :

`rpi-update && reboot`

reboots the pi even if no firmware wasn't updated.

I know that this can be handled by testing the output for the string "reboot required"... But a return code is better than checking for a specific string which can be subject to change ?

popcornmix, 2015-01-04:
Can you use JUST_CHECK from here?

mbocquet, 2015-01-04:
I will ! But do you plan to use return codes in the future?

thib3113, 2015-01-16:
i think to is a good idea, it's easier and faster to look at a number rather than a big sentence with lot of commits etc. Or you can make a no-verbose mode

mbocquet, 2015-02-07:
Here is a wrapper script for my needs...
[rnrpi-update](https://github.com/mbocquet/rnrpi-update)

Ruffio, 2016-04-30:
@popcornmix is something like this going to be implemented or should this issue be closed as wont fix?

Ruffio, 2016-05-20:
@popcornmix any thoughts?