Hexxeh/rpi-update #158: Does not display appropriate warnings or disclaimers before installing/updating

Gadgetoid, 2015-01-26:
This utility is widely available, easy to obtain and far too easily confused with an "official" script such as raspi-config. I suggest it includes a warning making it explicitly clear what it is and does before proceeding to upgrade the firmware to a potentially experimental, bleeding-edge release.

I've historically assumed this script was the go-to way to obtain new firmware releases, and thus recommended it in documentation. I've since retracted those recommendations, but feel the script itself doesn't do the due-diligence to ensure it's not run erroneously.

Recent support requests where users are seeing the i2c devices disappear upon updating are very telling that this is a genuine problem.

popcornmix, 2015-01-26:
To be honest the i2c devices disappearing is not caused by rpi-update, but to a necessary change to the firmware/kernel.

You will get exactly the same behaviour at some point in the future through apt-get upgrade or when installing a new raspbian image.

However I guess we could *sometimes* add a warning file to the firmware repo, which if present, the rpi-update tool will print out and ask "Are you sure?".

Currently that could include a link to http://www.raspberrypi.org/forums/viewtopic.php?f=29&t=97216

Gadgetoid, 2015-01-26:
I'm aware that the devices disappearing is not caused by rpi-update itself, but by progressive and necessary changes to the firmware it installs. I'm hoping that some sort of useful feedback, message or warning would be implemented for the users who seem to "rpi-update" without understanding fully what it is or does.

It will be largely my responsibility ( a large part of my role at Pimoroni is supporting hardware products with software ) to resolve the mess that occurs when these changes are rolled out to a wider audience, however, so these early support requests are useful.

NeonHorizon, 2015-01-26:
I agree with Gadgetoid, I've also had issues with people using this script and breaking compatibility with my project BerryIO. It would be nice if the script could warn people that they are moving their Pi to bleeding edge code and that this may cause problems with software and hardware.

I realise its not actually the fault of rpi-update itself which I'm sure we agree is an excellent tool, back in the day and it was pretty much required just to get your system up and running as things were moving at such a fast pace, but now we have a stable Raspbian image the average user should not really need to be using this and it can cause problems as developers have not finished testing their applications on systems with these configurations. 

The biggest problem is unfortunately many blogs are still erroneously recommending people run this tool as if its the normal process. Its gonna be almost impossible to get all those blogs changed but perhaps a simple warning could work wonders instead?

Ruffio, 2021-03-29:
Please consider to close this 'issue'. rpi-update tells the user that:

root@raspberrypi:~# rpi-update
 *** Raspberry Pi firmware updater by Hexxeh, enhanced by AndrewS and Dom
 *** Performing self-update
 *** Relaunching after update
 *** Raspberry Pi firmware updater by Hexxeh, enhanced by AndrewS and Dom
 *** We're running for the first time
 *** Backing up files (this will take a few minutes)
 *** Remove old firmware backup
 *** Backing up firmware
 *** Backing up modules 4.19.57+
#############################################################
WARNING: 'rpi-update' updates to pre-releases of the linux 
kernel tree and Videocore firmware.

'rpi-update' should only be used if there is a specific 
reason to do so - for example, a request by a Raspberry Pi 
engineer.

DO NOT use 'rpi-update' as part of a regular update process.

##############################################################
Would you like to proceed? (y/N)

Gadgetoid, 2021-03-29:
Agreed! This is ancient history now.

closed 2021-03-29 by Gadgetoid