Hexxeh/rpi-update #163: ipv6 support

kugelzucker, 2015-02-19:
hi,

i am running ipv6-only and encountered the following problem while running the following:

$ sudo rpi-update
 *** Raspberry Pi firmware updater by Hexxeh, enhanced by AndrewS and Dom
 *** Performing self-update
curl: (7) Failed to connect to 23.235.47.133: Network is unreachable
 !!! Failed to download update for rpi-update!
 !!! Make sure you have ca-certificates installed and that the time is set correctly

seems pretty straightforward for me ;) please support ipv6

merge, 2015-05-25:
please fix this. I have the same problem. Are there workarounds for the time being?

xrobin, 2015-06-05:
This looks like a github issue, as the script pulls from https://github.com and https://raw.githubusercontent.com. See github/github-services/issues/489.

popcornmix, 2015-06-05:
rpi-update just uses curl to update. i don't believe it contains anything ipv4 or ipv6 specific.
If curl is not working on your platform then I think that is something unrelated to rpi-update.

kugelzucker, 2015-06-11:
the platform is a current raspbian, but seeing that rpi-update is using github in the process i guess we know what needs to be fixed. github needs to finally step up with ipv6-support. thank you for responding.

rmyorston, 2016-02-29:
My Pi Zero is IPv6-only.  Happily, though, it's able to access a squid server on my firewall which _can_ talk to the old-fashioned parts of the interwebs.  So I'm able to use this as a workaround:
```
sudo ALL_PROXY=janus:3128 rpi-update
```
It's rather irritating that GitHub doesn't have IPv6 support.

Ruffio, 2016-07-24:
@kugelzucker please consider closing this issue. The issue can't be resolved by any of the members here as the issue is with Github itself and is not specificeret for this Gist. Please write to Github support so they know of the demand for this feature. Thanks.

closed 2016-07-25 by popcornmix