Hexxeh/rpi-update #165: Kernel panic??

tsamboni, 2015-03-12:
I have no idea what happened but after updating firmware my RPi doesnt power up.. I have error log here

user@pi ~ $ sudo rpi-update 
 *** Raspberry Pi firmware updater by Hexxeh, enhanced by AndrewS and Dom
 *** Performing self-update
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100  8107  100  8107    0     0  18523      0 --:--:-- --:--:-- --:--:-- 25413
 *** Relaunching after update
 *** Raspberry Pi firmware updater by Hexxeh, enhanced by AndrewS and Dom
 *** Downloading specific firmware revision (this will take a few minutes)
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100   168    0   168    0     0    228      0 --:--:-- --:--:-- --:--:--   323
100 43.5M  100 43.5M    0     0   448k      0  0:01:39  0:01:39 --:--:--  157k
 *** Updating firmware

Message from syslogd@raspberrypi at Mar 11 08:38:42 ...
 kernel:[150191.456842] Internal error: Oops: 17 [#1] PREEMPT ARM

Message from syslogd@raspberrypi at Mar 11 08:38:42 ...
 kernel:[150191.545864] Process mmcqd/0 (pid: 42, stack limit = 0xdb2941b0)

Message from syslogd@raspberrypi at Mar 11 08:38:42 ...
 kernel:[150191.551860] Stack: (0xdb295d90 to 0xdb296000)

Message from syslogd@raspberrypi at Mar 11 08:38:42 ...
 kernel:[150191.556300] 5d80:                                     00000000 c02f062c 00000000 00000001

Message from syslogd@raspberrypi at Mar 11 08:38:42 ...
 kernel:[150191.564558] 5da0: c043dfbc db1c9c00 db28702c db294010 60000013 db1c9e40 db287240 00000001

Message from syslogd@raspberrypi at Mar 11 08:38:42 ...
 kernel:[150191.572817] 5dc0: db295df4 db295dd0 c043ee48 c043d980 c043ed50 db28702c db1c9c00 db28712c

Message from syslogd@raspberrypi at Mar 11 08:38:42 ...
 kernel:[150191.581074] 5de0: 00000001 db295eac db295e0c db295df8 c0424f90 c043ed5c db1c9c00 00000000

Message from syslogd@raspberrypi at Mar 11 08:38:42 ...
 kernel:[150191.589331] 5e00: db295e5c db295e10 c0425f60 c0424ec8 c0434918 db287140 db295e4c 00000000

Message from syslogd@raspberrypi at Mar 11 08:38:42 ...
 kernel:[150191.597587] 5e20: db298000 c004fdc4 db295e28 db295e28 db28712c d0bbcbc8 db286c00 db286c00

Message from syslogd@raspberrypi at Mar 11 08:38:42 ...
 kernel:[150191.605843] 5e40: db28702c db287004 db286c00 24590001 db295edc db295e60 c04354f4 c0425c54

Message from syslogd@raspberrypi at Mar 11 08:38:42 ...
 kernel:[150191.614099] 5e60: 00000000 00000000 db295e94 db295e78 d0bbcbc8 d0bbcc18 db295e9c db295e88

Message from syslogd@raspberrypi at Mar 11 08:38:42 ...
 kernel:[150191.622355] 5e80: c02c6264 c02f90c0 db2a8ba0 db2a8ba4 db295ebc 00000000 00000000 db287000

Message from syslogd@raspberrypi at Mar 11 08:38:42 ...
 kernel:[150191.630612] 5ea0: 00000000 d0bbcbc8 c007ead4 c001d478 cd73eff8 db287004 db286c00 d0bbcbc8

Message from syslogd@raspberrypi at Mar 11 08:38:42 ...
 kernel:[150191.638868] 5ec0: db287000 db287000 db287004 24590001 db295f2c db295ee0 c04360f8 c0435430

Message from syslogd@raspberrypi at Mar 11 08:38:42 ...
 kernel:[150191.647124] 5ee0: 00000000 d0bbcbc8 00000000 24590001 db295f14 db295f00 c02ccee8 db1c9c00

Message from syslogd@raspberrypi at Mar 11 08:38:42 ...
 kernel:[150191.655380] 5f00: 00000000 db294030 00000000 24590001 db294030 db294000 db287004 d0bbcbc8

Message from syslogd@raspberrypi at Mar 11 08:38:42 ...
 kernel:[150191.663636] 5f20: db295f64 db295f30 c04374d8 c0436000 db28700c db292000 c04373e8 db2aa7a0

Message from syslogd@raspberrypi at Mar 11 08:38:42 ...
 kernel:[150191.671892] 5f40: 00000000 db287004 c04373e8 00000000 00000000 00000000 db295fac db295f68

Message from syslogd@raspberrypi at Mar 11 08:38:42 ...
 kernel:[150191.680147] 5f60: c003edf8 c04373f4 db295f94 00000000 c0046010 db287004 00000000 db295f7c

Message from syslogd@raspberrypi at Mar 11 08:38:42 ...
 kernel:[150191.688403] 5f80: db295f7c 00000000 db295f88 db295f88 db2aa7a0 c003ed20 00000000 00000000

Message from syslogd@raspberrypi at Mar 11 08:38:42 ...
 kernel:[150191.696658] 5fa0: 00000000 db295fb0 c000e988 c003ed2c 00000000 00000000 00000000 00000000

Message from syslogd@raspberrypi at Mar 11 08:38:42 ...
 kernel:[150191.704913] 5fc0: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000

Message from syslogd@raspberrypi at Mar 11 08:38:42 ...
 kernel:[150191.713168] 5fe0: 00000000 00000000 00000000 00000000 00000013 00000000 52c2f460 20d2e0c1

Message from syslogd@raspberrypi at Mar 11 08:38:42 ...
 kernel:[150191.788461] Code: 0a000029 e59530ac e596e000 e3a00003 (e5931028)

popcornmix, 2015-03-12:
Was this running raspbian? What kernel version were you on? Does it still boot?

tsamboni, 2015-03-12:
Yes, its raspbian. Kernel is very new, it's not booting so I cant check the version but I was apt-get dist-upgraded about few days ago.

popcornmix, 2015-03-12:
Are you overclocking?
You say it's not booting - how far does it get? Anything on display? 
You might want to compare the files on the boot partition (start.elf / kernel.img etc) with the ones here:
https://github.com/Hexxeh/rpi-firmware

If any are different, then download from github and replace. Keep a copy of the "broken" files as it may be interesting to see what the issue was.

tsamboni, 2015-03-12:
Yes, its little overclocked. Cant remember how much. Its not acceptable via ssh but now I added display and it bootet until 18.something number goes on. I replaced kernel.img and start* files and now its working again! I saved the overwrited files, do you want them somehow?

popcornmix, 2015-03-12:
It seems the Pi crashed during the update (possibly because the overclock is too high, an insufficient power supply, sdcard problems or some other bug) resulting in a corrupt/incomplete kernel/firmware. 

If you can upload the files somewhere (perhaps google drive or dropbox) it might be interesting to see what's wrong (e.g. if they are corrupted or truncated).

Ruffio, 2015-06-14:
@tsambni can you provide the files and can this issue be closed?

Ruffio, 2016-04-30:
@tsamboni can this be closed?

tsamboni, 2016-05-03:
Yes you can.

closed 2016-05-03 by popcornmix