Hexxeh/rpi-update #166: Panic VFS after update

matthuisman, 2015-03-20:
Using stock NOOBS
Boot into Raspbian.
Run sudo rpi-udapte
Updates successfully and says reboot.
Reboot.

Message "PANIC: VFS: Unable to mount root fs on unknown-lock(179,2)

This is the 2nd SD card I have tried.
Using very good power supply with thick DC cables. Only ethernet and keyboard plugged in.

It also "kills" NOOBS. Can't hold shift to get into NOOBS anymore. Straight into Raspbian boot then panic.

Almost like NOOBS has been corrupted by the rpi-update

matthuisman, 2015-03-20:
Have posted an issue on RPI firmware with more details here:
https://github.com/raspberrypi/firmware/issues/398

popcornmix, 2015-03-20:
Duplicate of raspberrypi/firmware#398

closed 2015-03-20 by popcornmix