Hexxeh/rpi-update #168: Random reboot after upgrade

ljcljc, 2015-03-20:
I run 'rpi-update' to update kernel to the latest version, and now it won't be dead, but it randomly reboot for no reason. Sometime it's one hour, sometimes it's 2 hours.

This is when I run 'last reboot'

reboot system boot 3.18.9-v7+ Fri Mar 20 15:46 - 17:54 (02:08)
reboot system boot 3.18.9-v7+ Fri Mar 20 14:17 - 15:46 (01:29)
reboot system boot 3.18.9-v7+ Fri Mar 20 14:03 - 15:46 (01:43)

pi would automatically reboot as I see in /var/log/syslog

Mar 20 15:45:48 205-1-1 /USR/SBIN/CRON[4215]: (CRON) info (No MTA installed, discarding output)
Mar 20 15:46:36 205-1-1 shutdown[4310]: shutting down for system reboot
Mar 20 15:46:36 205-1-1 init: Switching to runlevel: 6
Mar 20 15:46:39 205-1-1 mysqld: 150320 15:46:39 [Note] /usr/sbin/mysqld: Normal shutdown
Mar 20 15:46:39 205-1-1 mysqld:
Mar 20 15:46:39 205-1-1 mysqld: 150320 15:46:39 [Note] Event Scheduler: Purging the queue. 0 events
Mar 20 15:46:39 205-1-1 mysqld: 150320 15:46:39 InnoDB: Starting shutdown...
Mar 20 15:46:41 205-1-1 mysqld: 150320 15:46:41 InnoDB: Shutdown completed; log sequence number 718843536
Mar 20 15:46:41 205-1-1 mysqld: 150320 15:46:41 [Note] /usr/sbin/mysqld: Shutdown complete
Mar 20 15:46:41 205-1-1 mysqld:
Mar 20 15:46:42 205-1-1 mysqld_safe: mysqld from pid file /var/run/mysqld/mysqld.pid ended
Mar 20 15:46:47 205-1-1 kernel: imklog 5.8.11, log source = /proc/kmsg started.
Mar 20 15:46:47 205-1-1 rsyslogd: [origin software="rsyslogd" swVersion="5.8.11" x-pid="2010" x-info="http://www.rsyslog.com"] start
Mar 20 15:46:47 205-1-1 kernel: [ 0.000000] Booting Linux on physical CPU 0xf00
Mar 20 15:46:47 205-1-1 kernel: [ 0.000000] Initializing cgroup subsys cpu
Mar 20 15:46:47 205-1-1 kernel: [ 0.000000] Initializing cgroup subsys cpuacct

My pi is working perfectly for several months on rpi b+, and has problems after I upgrade. I believe it's a problem with kernel. Can someone help me with that? Thanks

popcornmix, 2015-03-20:
I can't see a reason for this being kernel related. Looks like a cron job.
Can you revert to an earlier firmware and confirm the problem goes away? Then we can identify the point where it occurred.

From: https://github.com/Hexxeh/rpi-firmware/commits/master
you can find the hash of any firmware update (the hex number at then end of the URL).

Downgrade with:
[code]sudo rpi-update <hash>[/code]

What kernel version were you on before.
Try reverting to a version close to that and check if the issue is present.

ljcljc, 2015-03-20:
My kernel version is 3.12.28 on pi b+, it's working fine.
Then, I upgrade to 3.18.7 for pi2, it will random be dead, I have to replug the power to reboot it manually.
Now, I am on 3.18.9, it will random reboot, not dead. At least I don't need to unplug and plug the power from time to time.

I cannot go back, because it won't support rpi2 with old version. And I am sure it's not a crob job problem. The only thing I do using cron is to take picture with a webcam, and that should not cause reboot. 

However, it has error message like this:
uvcvideo: Failed to query (GET_DEF) UVC control 5 on unit 3: -32 (exp. 1).

But I happened before, nothing happened, and the webcam is working fine. So I just ignore it.

Also, here is my dmesg output if you have interest to read. Thanks.

[ 0.000000] Booting Linux on physical CPU 0xf00
[ 0.000000] Initializing cgroup subsys cpu
[ 0.000000] Initializing cgroup subsys cpuacct
[ 0.000000] Linux version 3.18.9-v7+ (dc4@dc4-XPS13-9333) (gcc version 4.8.3 20140303 (prerelease) (crosstool-NG linaro-1.13.1+bzr2650 - Linaro GCC 2014.03) ) #768 SMP PREEMPT Sun Mar 15 19:41:56 GMT 2015
[ 0.000000] CPU: ARMv7 Processor [410fc075] revision 5 (ARMv7), cr=10c5387d
[ 0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT aliasing instruction cache
[ 0.000000] Machine model: Raspberry Pi 2 Model B
[ 0.000000] cma: Reserved 8 MiB at 0x36800000
[ 0.000000] Memory policy: Data cache writealloc
[ 0.000000] On node 0 totalpages: 225280
[ 0.000000] free_area_init_node: node 0, pgdat 80809440, node_mem_map b6113000
[ 0.000000] Normal zone: 1760 pages used for memmap
[ 0.000000] Normal zone: 0 pages reserved
[ 0.000000] Normal zone: 225280 pages, LIFO batch:31
[ 0.000000] [bcm2709_smp_init_cpus] enter (8620->f3003010)
[ 0.000000] [bcm2709_smp_init_cpus] ncores=4
[ 0.000000] PERCPU: Embedded 10 pages/cpu @b60e1000 s11456 r8192 d21312 u40960
[ 0.000000] pcpu-alloc: s11456 r8192 d21312 u40960 alloc=10*4096
[ 0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3
[ 0.000000] Built 1 zonelists in Zone order, mobility grouping on. Total pages: 223520
[ 0.000000] Kernel command line: dma.dmachans=0x7f35 bcm2708_fb.fbwidth=656 bcm2708_fb.fbheight=416 bcm2709.boardrev=0xa21041 bcm2709.serial=0x76b23710 smsc95xx.macaddr=B8:27:EB:B2:37:10 bcm2708_fb.fbswap=1 bcm
0000 vc_mem.mem_base=0x3dc00000 vc_mem.mem_size=0x3f000000 dwc_otg.lpm_enable=0 console=tty1 root=/dev/mmcblk0p2 rootfstype=ext4 elevator=deadline rootwait
[ 0.000000] PID hash table entries: 4096 (order: 2, 16384 bytes)
[ 0.000000] Dentry cache hash table entries: 131072 (order: 7, 524288 bytes)
[ 0.000000] Inode-cache hash table entries: 65536 (order: 6, 262144 bytes)
[ 0.000000] Memory: 875832K/901120K available (5705K kernel code, 397K rwdata, 1740K rodata, 384K init, 763K bss, 25288K reserved)
[ 0.000000] Virtual kernel memory layout:
[ 0.000000] vector : 0xffff0000 - 0xffff1000 ( 4 kB)
[ 0.000000] fixmap : 0xffc00000 - 0xffe00000 (2048 kB)
[ 0.000000] vmalloc : 0xb7800000 - 0xff000000 (1144 MB)
[ 0.000000] lowmem : 0x80000000 - 0xb7000000 ( 880 MB)
[ 0.000000] modules : 0x7f000000 - 0x80000000 ( 16 MB)
[ 0.000000] .text : 0x80008000 - 0x8074d700 (7446 kB)
[ 0.000000] .init : 0x8074e000 - 0x807ae000 ( 384 kB)
[ 0.000000] .data : 0x807ae000 - 0x8081147c ( 398 kB)
[ 0.000000] .bss : 0x8081147c - 0x808d01d4 ( 764 kB)
[ 0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[ 0.000000] Preemptible hierarchical RCU implementation.
[ 0.000000] NR_IRQS:608
[ 0.000000] Architected cp15 timer(s) running at 19.20MHz (virt).
[ 0.000013] sched_clock: 56 bits at 19MHz, resolution 52ns, wraps every 3579139424256ns
[ 0.000037] Switching to timer-based delay loop, resolution 52ns
[ 0.000339] Console: colour dummy device 80x30
[ 0.001795] console [tty1] enabled
[ 0.001845] Calibrating delay loop (skipped), value calculated using timer frequency.. 38.40 BogoMIPS (lpj=192000)
[ 0.001929] pid_max: default: 32768 minimum: 301
[ 0.002332] Mount-cache hash table entries: 2048 (order: 1, 8192 bytes)
[ 0.002393] Mountpoint-cache hash table entries: 2048 (order: 1, 8192 bytes)
[ 0.003671] Initializing cgroup subsys memory
[ 0.003762] Initializing cgroup subsys devices
[ 0.003817] Initializing cgroup subsys freezer
[ 0.003870] Initializing cgroup subsys net_cls
[ 0.003937] Initializing cgroup subsys blkio
[ 0.004048] CPU: Testing write buffer coherency: ok
[ 0.004161] ftrace: allocating 19576 entries in 58 pages
[ 0.052398] CPU0: update cpu_capacity 1024
[ 0.052478] CPU0: thread -1, cpu 0, socket 15, mpidr 80000f00
[... 196 lines cut ...]
[ 1275.053744] uvcvideo: Failed to query (GET_DEF) UVC control 5 on unit 3: -32 (exp. 1).
[ 1453.708466] uvcvideo: Failed to query (GET_DEF) UVC control 5 on unit 3: -32 (exp. 1).
[ 1808.639728] uvcvideo: Failed to query (GET_DEF) UVC control 5 on unit 3: -32 (exp. 1).
[ 2059.469137] uvcvideo: Failed to query (GET_DEF) UVC control 5 on unit 3: -32 (exp. 1).
[ 2427.624575] uvcvideo: Failed to query (GET_DEF) UVC control 5 on unit 3: -32 (exp. 1).
[ 2801.499533] uvcvideo: Failed to query (GET_DEF) UVC control 5 on unit 3: -32 (exp. 1).
[ 3079.373394] uvcvideo: Failed to query (GET_DEF) UVC control 5 on unit 3: -32 (exp. 1).
[ 3081.085504] uvcvideo: Failed to resubmit video URB (-1).
[ 3254.291254] uvcvideo: Failed to query (GET_DEF) UVC control 5 on unit 3: -32 (exp. 1).
[ 3627.569133] uvcvideo: Failed to query (GET_DEF) UVC control 5 on unit 3: -32 (exp. 1).
[ 3629.252706] uvcvideo: Failed to resubmit video URB (-1).
[... cut ...]
[ 6330.229578] uvcvideo: Failed to query (GET_DEF) UVC control 5 on unit 3: -32 (exp. 1).
[ 6578.240840] uvcvideo: Failed to query (GET_DEF) UVC control 5 on unit 3: -32 (exp. 1).
[ 6855.174948] uvcvideo: Failed to query (GET_DEF) UVC control 5 on unit 3: -32 (exp. 1).
[ 6856.973911] uvcvideo: Failed to resubmit video URB (-1).
[ 7146.162878] uvcvideo: Failed to query (GET_DEF) UVC control 5 on unit 3: -32 (exp. 1).
[ 7472.168087] uvcvideo: Failed to query (GET_DEF) UVC control 5 on unit 3: -32 (exp. 1).

popcornmix, 2015-03-20:
Does it reboot when you access the camera? Might be a power supply issue and enabling the camera causes a voltage drop and reboots the Pi.

Can you try a different power supply or measure the voltage when running with camera working?

ljcljc, 2015-03-21:
No. I took camera every 5 minutes. It's usually fine. I will try a different power.

And do you know at what power voltage it will reboot? What's the required power range for pi2?

popcornmix, 2015-03-21:
Pi needs 5V +/- 5%
You need to measure the voltage when under load (e.g. with camera running).

Ruffio, 2015-06-14:
@ljcljc is this still an issue?

ljcljc, 2015-06-15:
Yes. I had to switch back to old B+ version of Pi, and there is no problem
then.

popcornmix, 2015-06-15:
Is the B+ happy with latest kernel?
I'm sure this is a power supply issue.

ljcljc, 2015-06-15:
Yes. B+ is stable without any problem. I have tried many different power
with RPI 2, but none of them work, after then I just gave up.

Ruffio, 2016-04-30:
@ljcljc can this be closed or how do we her any further? I'm thinking that the issue is/was locally else we would have seen a lot of people writting in hear.

Ruffio, 2016-05-20:
@ljcljc is this still an issue? I'm thinking that the issue is/was locally else we would have seen a lot of people writting in here... Please consider closing this issue. Thanks.

ljcljc, 2016-05-22:
ok, I will close it

closed 2016-05-22 by ljcljc