Hexxeh/rpi-update #169: Does not remove old firmware backup directory and does not backup modules.

RagnarJensen, 2015-03-26:
In function do_backup 
`if [[ -f "${FW_PATH}.bak" ]]`
and 
`if [[ -f "${FW_MODPATH}/$(uname -r)" ]]`

The -f flag makes them look for files, when they should look for directories using the -d flag.
The directories in question are thus not removed and backed up, respectively.

popcornmix, 2015-03-26:
Have you tested with the -f changed to -d? Did it work as desired?

RagnarJensen, 2015-03-26:
Yes, I edited rpi-update to test and it works with the -d flag.
/boot.bak is removed and my modules are backed up.

Quick test:
`root@quadripi:~# ls -ld /boot*`
`drwxr-xr-x 3 root root 16384 Jan  1  1970 /boot`
`drwxr-xr-x 4 root root  4096 Feb 28 20:03 /boot.bak`

`root@quadripi:~# FW_PATH=/boot`
`root@quadripi:~# if [[ -f "${FW_PATH}.bak" ]]; then echo found a directory; fi`
`root@quadripi:~#`
`root@quadripi:~# if [[ -d "${FW_PATH}.bak" ]]; then echo found a directory; fi`
`found a directory`

popcornmix, 2015-03-26:
There seemed to be a third instance where -f was used on a directory. I've updated - can you test?

RagnarJensen, 2015-03-26:
Yes, I noticed that too.
I've downloaded your update, it tests OK.

RagnarJensen, 2015-03-26:
Tested on a couple more Pis. OK.

closed 2015-03-26 by RagnarJensen

popcornmix, 2015-03-26:
Cool. Thanks for report.