Hexxeh/rpi-update #172: Multiple 1-Wire busses?

phmarek, 2015-05-03:
I've got two DS18B20 in the immediate vicinity of my RPi, so querying them via GPIO works fine.

Then I've got 5 more, but these are located about 15m further (and are then chained up one after another on a cable of ~2m length); this doesn't work with the GPIO port anymore, although I already have Vcc on 5V and tried 4k7 to 1k5 for the pullup (against 3V3, of course).

So, my next try would be to use a serial-to-1-wire IC after the 15m and before the 2m cables; but that would mean that the RPi would need to either show 2 1-wire busses, or that they need to get accumulated into the one directory.

Is that possible with the upstream kernel already? ISTR that there have been patches floating around for that. Or, is there another way to get stable readings?

kmitchel, 2015-07-11:
Don't use parasitic mode.

phmarek, 2015-07-11:
I'm not. I've even had them powered via +5V, but it still didn't work across that distance with that cable.

popcornmix, 2015-07-12:
Closing as this is the wrong place. (This is repo for script which updates the kernel).
The forum is the best place for support questions like this:
https://www.raspberrypi.org/forums/
If you have a confirmed kernel but then try here:
https://github.com/raspberrypi/linux/issues

closed 2015-07-12 by popcornmix