Hexxeh/rpi-update #173: RPI2 Raspbian: rpi-update =>  /proc/config.gz changed => SELinux disabled

RamiTITG, 2015-05-12:
Hello guys, question: I compiled the Raspbian kernel in Raspberry Pi 2 to enable all the SELinux parameters in .config, and it worked, SELinux was activated with the new /boot/kernel7.img

However, I didn't realise rpi-update updates the kernel7.img, and after running rpi-update, SELinux was not enabled anymore. I could see in /proc/config.gz that all SELinux parameters are not there anymore (CONFIG_SECURITY_SELINUX=y, etc......)

The situation is not practical as locally compiling in RP2 takes loads of hours, and I'm lazy enough to cross-compile, so I can't use SELinux if the kernel stops supporting it every time I run rpi-update.

Is there any way to run rpi-update to keep those SELinux parameters (/proc/config.gz) or any other thing you could think of to avoid losing SELinux support in the kernel after running rpi-update? (maybe not running rpi-update and do something else?), or maybe I'm completely lost and you see this from a different angle?

Thanks and cheers
Ramiro.

popcornmix, 2015-05-12:
If you run a custom kernel, then you shouldn't run rpi-update.
There is no way to "merge" the changes in your kernel with the changes we have made.

If you just want to update the GPU and leave your kernel intact then run:
[code]sudo SKIP_KERNEL=1 rpi-update[/code]

RamiTITG, 2015-05-12:
Solid, I appreciate your quick response. I'll use that parameter.

Cheers
Ramiro

closed 2015-05-12 by RamiTITG