Hexxeh/rpi-update #174: rpi-update not finishing downloads

risingsunomi, 2015-05-20:
Hello,

When I run rpi-update it doesn't seem to be finishing the downloads for the updates it is doing. I have everything installed for rpi-update but it seems that the update is not finishing still. This is on an rpi2

 *** Raspberry Pi firmware updater by Hexxeh, enhanced by AndrewS and Dom
 *** Performing self-update
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100  9823  100  9823    0     0  39746      0 --:--:-- --:--:-- --:--:-- 54572
 *** Relaunching after update
 *** Raspberry Pi firmware updater by Hexxeh, enhanced by AndrewS and Dom
 *** We're running for the first time
 *** Backing up files (this will take a few minutes)
 *** Remove old firmware backup
 *** Backing up firmware
 *** Remove old modules backup
 *** Backing up modules 3.18.11-v7+
 *** Downloading specific firmware revision (this will take a few minutes)
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100   168    0   168    0     0   1201      0 --:--:-- --:--:-- --:--:--  1647
 24 46.2M   24 11.2M    0     0   998k      0  0:00:47  0:00:11  0:00:36  682k

gzip: stdin: unexpected end of file
tar: Unexpected EOF in archive
tar: Unexpected EOF in archive
tar: Error is not recoverable: exiting now

popcornmix, 2015-05-20:
Wired or wireless?
Do you have any other network issues?
Does it sometimes work? (i.e. it could be temporary issues at the other end).

I've ran rpi-update a number of times today without issue, so this isn't a general problem.

risingsunomi, 2015-05-21:
It was wired but I think the place I was at was having network issues. Was disconnecting on other machines I noticed. Going to try again today on my home connection and see if it goes through. 

Thanks

Ruffio, 2015-06-14:
@risingsunomi do you still experience this issue? It looks like it was locsl network issues. Can this issue be closed?

risingsunomi, 2015-07-15:
This can be closed and the issue was fixed. Was a local network issue.

closed 2015-07-15 by risingsunomi

maraouf, 2016-04-29:
I'm facing the same issue now, i am connected to wired connection and was updating normally via rpi-update but after installing the release 
version 9dc453dc69db79656c9920742f6d51c604356a8b (clean) (release)
Linux raspberrypi 4.4.7+ #878 Tue Apr 19 18:59:39 BST 2016 armv6l GNU/Linux

I started experiencing this issue

KthProg, 2017-11-29:
Same issue. Not sure at all what the issue is.

KthProg, 2017-11-29:
All good now, there was not enough space on the pi it looks like.

mridulweb, 2018-03-17:
I had the same issue for about a month when I realized the only new thing I enabled on my pi was 'motion' for motion detection pictures/videos and some NFS mounts. I did notice the CPU high most of the times due to motion. So I stopped motion, disabled NFS mounts and ran rpi-update and IT WORKED LIKE A CHARM IN ONE SHOT!

mridulweb, 2018-03-17:
Also, my pi3 is hooked up to a pi3 power supply, connected via gigabit network hard-wired (USB to Ethernet to achive gigabit speed over home LAN), and space was never an issue

anberra, 2018-03-22:
I just resolv this issue on my pi zero w it by creating file ~root/.curlrc containing the following line
--keepalive-time 2

chuckenglehart, 2018-03-26:
> I just resolv this issue on my pi zero w it by creating file ~root/.curlrc containing the following line
--keepalive-time 2

Thank you, that fixed the issue for me.

DolleDries, 2018-04-03:
Weird. The default value for keepalive-time is 60 (seconds).... Still having issues after altering it ot 120...
        curl: (18) transfer closed with 28876575 bytes remaining to read

popcornmix, 2018-04-03:
The suggestion was to set it to 2, lower than the default of 60.
Why are you trying 120?

DolleDries, 2018-04-04:
> The suggestion was to set it to 2, lower than the default of 60.
> Why are you trying 120?
>

Because setting it to 2 didn't work either... (nor did 120 or 1)

--
               Andre van der Vlies <andre@vandervlies.xs4all.nl>
               Certifiable Linux/UNIX engineer (CLUE)
               Homepage: http://vandervlies.xs4all.nl/~andre
--
    'Reality is that which, when you stop believing in it, doesn't go away.'
                    -  Philip K. Dick
    # vim: ts=4 sw=4 ff=unix sm ai ic:

    ^[^#]
--

hobbsy, 2019-05-04:
> I just resolv this issue on my pi zero w it by creating file ~root/.curlrc containing the following line
> --keepalive-time 2

thank you @anberra - this worked for me on my Pi Zero W

For any beginners:

In the terminal change your current directory to the root directory of your pi
eg. starting from your home directory of /home/pi/ enter:

`cd ../..`

(you can check where you are by typing pwd)

then create a new file in that root folder called .curlrc, by using the text editor nano:

`sudo nano ./.curlrc`

type the line

`--keepalive-time 2`

then exit nano and save via Ctrl + X, and hit yes to save changes 

then try to run the command again

`sudo rpi-update`

HiranChaudhuri, 2019-06-19:
The suggestion did not work for me:

```
pi@garage:~ $ sudo su -
root@garage:~# echo --keepalive-time 2 >/.curlrc
root@garage:~# logout
pi@garage:~ $ cat /.curlrc 
--keepalive-time 2
pi@garage:~ $ sudo rpi-update
 *** Raspberry Pi firmware updater by Hexxeh, enhanced by AndrewS and Dom
 *** Performing self-update
 *** Relaunching after update
 *** Raspberry Pi firmware updater by Hexxeh, enhanced by AndrewS and Dom
 *** We're running for the first time
 *** Backing up files (this will take a few minutes)
 *** Remove old firmware backup
 *** Backing up firmware
 *** Remove old modules backup
 *** Backing up modules 4.19.50+
Connection to 192.168.178.49 closed by remote host.
Connection to 192.168.178.49 closed.
$ 
```

Rohlik, 2022-06-14:
Same error, setting `--keepalive-time 2` didn't help 😢.

popcornmix, 2022-06-15:
Note the original suggestion (that worked for some) was to edit the file `~root/.curlrc` which expands to `/root/.curlrc`
Some of the subsequent posts of it not working are using `/.curlrc` which won't have any effect.