Hexxeh/rpi-update #176: Alpine Linux compatibility

Diftraku, 2015-05-28:
Currently, when I try to run the script in Alpine Linux (alpine-rpi-3.2.0-armhf), BusyBox's _tar_ will complain about unrecognized option:
```tar: unrecognized option: strip-components=1```

So far, the following depencies are required to run the updater on Alpine:
* curl
* bash (BusyBox ash _may_ or _may not_ suffice)
* binutils
* tar

```apk add curl bash binutils tar```
After installing said depencies, the updater chucks along, filling the rootfs of the Alpine and eventually tar is unable to extract any modules into _/lib/modules_.

I am currently trying to run the updater with ```# ROOT_PATH=/ BOOT_PATH=/media/mmcblk0p1/boot rpi-update```

The problem is, _/lib/modules_ is a symlink to /.modloop/modules, which is the rootfs. Thus the space is very limited. _/media/mmcblk0p1_ is the SD card root that contains the _/boot_ on Alpine along with all the other OS related files. All of these are on a FAT32 partition.

popcornmix, 2015-05-29:
rpi-update is designed for raspbian. If it doesn't work on other distributions, then that is not something we will actively support.
You could set env variable "RPI_UPDATE_UNSUPPORTED=1" to indicate Alpine linux is unsupported.
Or you could submit a PR to change rpi-update. If it doesn't harm rpi-update on raspbian, then we'll consider accepting it.

Ruffio, 2015-06-21:
@Diftraku are you looking into making a PR or should this issue be closed?

Diftraku, 2015-06-21:
@Ruffio I haven't really looked into this at all lately. This issue can be closed for now, I'll reopen it or post a new one with the PR when I get around poking at my RPi.

closed 2015-06-21 by Diftraku