Hexxeh/rpi-update #177: Ability to suppress notice WARNING

jayslovak, 2015-06-22:
I am using rpi-update in an automated build script to prepare new raspberry images. The automated functionality is now broken because there is a warning in the NOTICE.md file that requires user input:
```
#############################################################
WARNING: This update bumps to rpi-4.0.y linux tree
Be aware there could be compatibility issues with some drivers
Discussion here:
https://www.raspberrypi.org/forums/viewtopic.php?f=29&t=113753
##############################################################
Would you like to proceed? (y/N)
```
Generated by function:
```
function show_notice {
local NOTICE=`curl -Lfs https://raw.githubusercontent.com/hexxeh/rpi-firmware/${FW_REV}/NOTICE.md`
if [ -z "$NOTICE" ]; then
return
fi
echo "$NOTICE"
if ! echo "$NOTICE" | grep -q WARNING; then
return
fi
read -p "Would you like to proceed? (y/N)" -n 1 -r -s
echo ""
if ! [[ $REPLY =~ ^[Yy]$ ]]; then
exit 1;
fi
}
```

Could we add a variable or a command line switch to suppress such notices?

Thanks

popcornmix, 2015-06-22:
Can you try now with:
``sudo SKIP_WARNING=1 rpi-update``

jayslovak, 2015-06-22:
Now it works EXACTLY as I needed.

No prompt when variable set:
```
#############################################################
WARNING: This update bumps to rpi-4.0.y linux tree
Be aware there could be compatibility issues with some drivers
Discussion here:
https://www.raspberrypi.org/forums/viewtopic.php?f=29&t=113753
##############################################################
 *** Downloading specific firmware revision (this will take a few minutes)
```

Thanks!

closed 2015-06-22 by popcornmix