Hexxeh/rpi-update #178: Don't display 4.0.y kernel warning if we're already on 4.0.y

shrx, 2015-06-23:
I upgraded my distros with `rpi-update` to the newest kernel:
```
uname -a
Linux raspberrypi2 4.0.5+ #797 PREEMPT Sat Jun 20 00:47:38 BST 2015 armv6l GNU/Linux
```
However, when I run  `rpi-update` again, it still warns me about upgrading:
```
sudo rpi-update
 *** Raspberry Pi firmware updater by Hexxeh, enhanced by AndrewS and Dom
 *** Performing self-update
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100 10185  100 10185    0     0  43123      0 --:--:-- --:--:-- --:--:-- 51964
 *** Relaunching after update
 *** Raspberry Pi firmware updater by Hexxeh, enhanced by AndrewS and Dom
#############################################################
WARNING: This update bumps to rpi-4.0.y linux tree
Be aware there could be compatibility issues with some drivers
Discussion here:
https://www.raspberrypi.org/forums/viewtopic.php?f=29&t=113753
##############################################################
```
The warning should not be issued anymore.

popcornmix, 2015-06-23:
That is not currently supported. The intention is the warning will be removed in a forthcoming update once we are happy there aren't any significant issues with the new kernel.

You can set SKIP_WARNING=1 to avoid the requirement to acknowledge this.

shrx, 2015-06-23:
Whoops, didn't realize that setting `SKIP_WARNING=1` doesn't actually remove the warning message. Should work now.

Ruffio, 2016-05-20:
@popcornmix what is the status on this one? The issue was raised june 2015 and I guess 'we' are happy with the new kernel?

commit 234c16ec (2016-07-24): check kernel version and issue warning if necessary

Fixes #178
closed 2016-07-24 by popcornmix