Hexxeh/rpi-update #181: Feature request: -Y option to force update if any available

Ruffio, 2015-07-16:
I'm building a web interface to control the RPi and it would be nice if there could be added a -y option, so if there are any updates available, the update is performed automatically.

I have tried various options but cant find options or man pages that can tell me if such an option exists.

tundra-git, 2015-07-17:
I also very much want this ability.   I want to be able to execute this program under automated workflows.  Forcing user interfaction will break such tool (tsshbatch, capistrano, ansible, rundeck ...)

tundra-git, 2015-07-27:
It seems that doing this first, gives desired behavior:

   export SKIP_WARNING=1

Is this a new feature or was it there and I just missed it in the doc somehow?

Ruffio, 2015-07-27:
How exactly do you do this? The other options are set like: sudo XXXX
rpi-update

popcornmix, 2015-07-27:
```sudo SKIP_WARNING=1 rpi-update```
would also work (they both just set a env variable).

Ruffio, 2015-07-28:
Thanks, this works perfectly :-)

It should be documented so no one is in doubt.

closed 2015-07-28 by Ruffio