Hexxeh/rpi-update #182: Please Don't Write To stderr With The Progress Report

tundra-git, 2015-07-17:
I am the author of the tsshbatch automation tool.  Your use of stderr for normal output is causing bad breakage.  Tools like tsshbatch remotely invoke things like this:

sudo rpi-update

When the sudo returns with anything in stderror, they assume that either the command itself or the sudo promotion failed.  That's what stderr is for - to report failures.  By having your progress report written to stderr, it's causing the automation workflow to break thinking something failed.

Why should you care?  Because people are going to want to be able to - say - automate the updating of every RPi in a lab without having to log into each one manually.

See more here:

https://ww.tundraware.com/Software/tsshbatch

Happy to help with code changes or testing.

Thanks!

popcornmix, 2015-07-25:
What are the stderr messages?

tundra-git, 2015-07-27:
Update download progress appears to be writing to stderr.  I've now worked around this  by invoking the program with 2>&1 appended to the command line.

popcornmix, 2015-07-27:
I believe that is just what curl does - I don't believe rpi-update is writing to stderr itself.
I think ``2>&1`` is the correct solution for you.

closed 2015-07-28 by tundra-git