Hexxeh/rpi-update #184: Not upgrading kernel version on Ubuntu mate

tolgamyth, 2015-07-22:
it says it is upgrading my kernel but uname -a command gives old kernel version, however i can see the latest kernel version in the specified folder.

tundra-git, 2015-07-22:
Not to be pedantic, but have you rebooted?

popcornmix, 2015-07-22:
rpi-update is designed for raspbian. It will only work on other systems if they behave similarly.
Is /dev/mmcblk0p0 mounted at /boot (and containing start.elf etc).

tolgamyth, 2015-07-22:
yes I have rebooted several times but I have expected it behaves same on ubuntu-MATE since both are debian based, maybe something act like bug on ubuntu-MATE.

tundra-git, 2015-07-22:
@tolgamyth - Ooops, I didn't spot that you were trying to do this on Ubuntu.  rpi-update is specific to RPi and not an Ubuntu tool.

You'll want to look into:  

  apt-get update
  apt-get upgrade
  apt-get dist-upgrade

tundra-git, 2015-07-22:
@popcornmix - Yep, I missed that the OP was trying this on Ubuntu.   Reading is hard :(

closed 2015-09-29 by popcornmix