Hexxeh/rpi-update #186: [New Feature Request]  Auto Reboot On Upgrade

tundra-git, 2015-07-28:
Again, in the spirit of supporting automated workflows, it would be nice to have an option to kick off a reboot automatically if the firmware gets upgraded.

popcornmix, 2015-07-29:
Anything wrong with:
```sudo rpi-update && sudo reboot```

tundra-git, 2015-07-29:
Does rpi-update set the error level such that this would not reboot if there were no firmware update?

tundra-git, 2015-07-29:
I just checked:  rpi-update always sets exit code to 0 regardless of whether there was a firmware update or not, so the above solution will ALWAYS reboot.  What I want is reboot only if there has been a firmware update.

Ruffio, 2016-04-30:
Duplicate of https://github.com/Hexxeh/rpi-update/issues/157 ?

tundratim, 2016-04-30:
Yes, they appear to be essentially identical issues.

closed 2016-11-22 by tundratim