Hexxeh/rpi-update #187: 8192cu drivers breaks WLAN connection often, bec. of new kernel 4.0?

ThomasKorimort, 2015-08-04:
Hi!
Since some time my WLAN connection which used to work properly apart from speed with EDIMAX WIFI dongle with RTL8188CUS chipset using RPI 2 B firmware driver 8192cu breaks too often to be useful now. I think this is maybe due to a firmware update. There was warning that the update could make some drivers not work anymore (kernel 4.0). Is this such a case? Further: Why do you provide an update, if some drivers don't work anymore afterwards?
I hope this issue can be solved nicely soon.

Many greetings, Thomas Korimort.

popcornmix, 2015-08-04:
This repo is for the rpi-update script. If you have a linux kernel issue then report here:
https://github.com/raspberrypi/linux/issues
where the right people are more likely to see it.

closed 2015-08-04 by popcornmix

ThomasKorimort, 2015-08-04:
thanx. i posted there.