Hexxeh/rpi-update #190: error running rpi-update curl: (6) Couldn't resolve host 'github.com'

TheQuestor, 2015-09-03:
When running rpi-update I now get an error that it can't resolve host github.com
 *** Downloading specific firmware revision (this will take a few minutes)
curl: (6) Couldn't resolve host 'github.com'

gzip: stdin: unexpected end of file
tar: Child returned status 1
tar: Error is not recoverable: exiting now

but I am able to ping it just fine

PING github.com (192.30.252.128) 56(84) bytes of data.
64 bytes from github.com (192.30.252.128): icmp_req=1 ttl=49 time=54.3 ms
64 bytes from github.com (192.30.252.128): icmp_req=2 ttl=49 time=53.9 ms
64 bytes from github.com (192.30.252.128): icmp_req=3 ttl=49 time=77.1 ms
64 bytes from github.com (192.30.252.128): icmp_req=4 ttl=49 time=51.2 ms

popcornmix, 2015-09-03:
Can you run:
``curl -L https://github.com/Hexxeh/rpi-firmware/archive/master.tar.gz > /dev/null``

and report the output?

Ruffio, 2016-04-30:
@TheQuestor is this still an issue? If yes please back with @popcornmix suggestion or close this issue if it is not relevant anymore.

TheQuestor, 2016-04-30:
It is working now but oddly not until I did a "sudo apt-get install ca-certificates" but yah I just did a "sudo rpi-update" and it all went through just fine but now my spi is no longer working sigh. and in X the taskbar keeps coming on and going off. The 5" TFT + Touch screen is working ok but it is all messed up. I should not have done this without backing up first :(

here is a video of what it is doing now.
https://youtu.be/TviZnNV8oys

going to try a dist-upgrade to see if it fixes it.

Ruffio, 2016-04-30:
@TheQuestor As the original issue is solved this should be closed then

TheQuestor, 2016-04-30:
Here is what I did 

pi@rpi2 /devel $ curl -L https://github.com/Hexxeh/rpi-firmware/archive/master.tar.gz
curl: (60) SSL certificate problem: unable to get local issuer certificate
More details here: http://curl.haxx.se/docs/sslcerts.html

curl performs SSL certificate verification by default, using a "bundle"
 of Certificate Authority (CA) public keys (CA certs). If the default
 bundle file isn't adequate, you can specify an alternate file
 using the --cacert option.
If this HTTPS server uses a certificate signed by a CA represented in
 the bundle, the certificate verification probably failed due to a
 problem with the certificate (it might be expired, or the name might
 not match the domain name in the URL).
If you'd like to turn off curl's verification of the certificate, use
 the -k (or --insecure) option.
pi@rpi2 /devel $ git config --global http.sslVerify true
pi@rpi2 /devel $ sudo apt-get install ca-certificates

pi@rpi2 /devel $ sudo rpi-update
 *** Raspberry Pi firmware updater by Hexxeh, enhanced by AndrewS and Dom
 *** Performing self-update
 *** Relaunching after update
 *** Raspberry Pi firmware updater by Hexxeh, enhanced by AndrewS and Dom
#############################################################
WARNING: This update bumps to rpi-4.4.y linux tree
Be aware there could be compatibility issues with some drivers
Discussion here:
https://www.raspberrypi.org/forums/viewtopic.php?f=29&t=144087
##############################################################
Would you like to proceed? (y/N)
 *** Downloading specific firmware revision (this will take a few minutes)
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100   168    0   168    0     0    493      0 --:--:-- --:--:-- --:--:--   721
100 50.4M  100 50.4M    0     0   862k      0  0:00:59  0:00:59 --:--:--  208k
 *** Updating firmware
 *** Updating kernel modules
 *** depmod 4.4.8-v7+
 *** depmod 4.4.8+
 *** Updating VideoCore libraries
 *** Using HardFP libraries
 *** Updating SDK
 *** Running ldconfig
 *** Storing current firmware revision
 *** Deleting downloaded files
 *** Syncing changes to disk
 *** If no errors appeared, your firmware was successfully updated to 384626a4e08d49262fd0ebb799b82f330ca7a0df
 *** A reboot is needed to activate the new firmware

pi@rpi2 /boot $ sudo reboot

and whilst the original issue seems to be fixed the result is far from good :(

but as requested I will close this and format my SD Card and start over from scratch [not happy sigh]

just for reference here is my dmesg

`pi@rpi2 ~ $ dmesg
[    0.000000] Booting Linux on physical CPU 0xf00
[    0.000000] Initializing cgroup subsys cpuset
[    0.000000] Initializing cgroup subsys cpu
[    0.000000] Initializing cgroup subsys cpuacct
[    0.000000] Linux version 4.4.8-v7+ (dc4@dc4-XPS13-9333) (gcc version 4.9.3 (crosstool-NG crosstool-ng-1.22.0-88-g8460611) ) #880 SMP Fri Apr 22 21:55:04 BST 2016
[    0.000000] CPU: ARMv7 Processor [410fc075] revision 5 (ARMv7), cr=10c5387d
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT aliasing instruction cache
[    0.000000] Machine model: Raspberry Pi 2 Model B Rev 1.1
[    0.000000] cma: Reserved 8 MiB at 0x36800000
[    0.000000] Memory policy: Data cache writealloc
[    0.000000] On node 0 totalpages: 225280
[... 235 lines cut ...]
[  161.299661] w1_master_driver w1_bus_master1: Family 0 for 00.c00000000000.ca is not registered.
[  232.889401] w1_master_driver w1_bus_master1: Family 0 for 00.200000000000.23 is not registered.
[  292.470707] w1_master_driver w1_bus_master1: Family 0 for 00.a00000000000.af is not registered.
[  327.957047] w1_master_driver w1_bus_master1: Family 0 for 00.600000000000.65 is not registered.
[  365.383779] w1_master_driver w1_bus_master1: Family 0 for 00.e00000000000.e9 is not registered.
[  424.451079] w1_master_driver w1_bus_master1: Family 0 for 00.100000000000.9d is not registered.
[  460.905545] w1_master_driver w1_bus_master1: Family 0 for 00.900000000000.11 is not registered.
[  531.571064] w1_master_driver w1_bus_master1: Family 0 for 00.500000000000.db is not registered.
[  580.012130] w1_master_driver w1_bus_master1: Family 0 for 00.d00000000000.57 is not registered.
[  604.621303] w1_master_driver w1_bus_master1: Family 0 for 00.300000000000.be is not registered.
[  651.181175] w1_master_driver w1_bus_master1: Family 0 for 00.b00000000000.32 is not registered.
[  687.921378] w1_master_driver w1_bus_master1: Family 0 for 00.700000000000.f8 is not registered.
[  736.345572] w1_master_driver w1_bus_master1: Family 0 for 00.f00000000000.74 is not registered.
[  794.843581] w1_master_driver w1_bus_master1: Family 0 for 00.080000000000.c2 is not registered.
[  844.152264] w1_master_driver w1_bus_master1: Family 0 for 00.880000000000.4e is not registered.
[  879.575110] w1_master_driver w1_bus_master1: Family 0 for 00.480000000000.84 is not registered.
[  927.002598] w1_master_driver w1_bus_master1: Family 0 for 00.c80000000000.08 is not registered.
[  998.594209] w1_master_driver w1_bus_master1: Family 0 for 00.280000000000.e1 is not registered.
[ 1057.993387] w1_master_driver w1_bus_master1: Family 0 for 00.a80000000000.6d is not registered.
[ 1117.520889] w1_master_driver w1_bus_master1: Family 0 for 00.680000000000.a7 is not registered.
[ 1154.964359] w1_master_driver w1_bus_master1: Family 0 for 00.e80000000000.2b is not registered.
[ 1214.104342] w1_master_driver w1_bus_master1: Family 0 for 00.180000000000.5f is not registered.
`

how can I "undo" the rpi-update?

closed 2016-04-30 by TheQuestor