Hexxeh/rpi-update #192: libkmod: ERROR. Pointing to previous kernel.

Fourdee, 2015-09-18:
Hi,

Ran rpi-update. It upgraded to 4.1.7.
Removed the previous kernels (4.1.6).

No sound, libkmod pointing to previous kernel.
> root@dietpi:/#  modprobe snd_bcm2835
libkmod: ERROR ../libkmod/libkmod.c:554 kmod_search_moddep: could not open moddep file '/lib/modules/4.1.6-v7+/modules.dep.bin'
root@dietpi:/# ls -lha /lib/modules/
total 16K
drwxr-xr-x  4 root root 4.0K Sep 18 12:34 .
drwxr-xr-x 13 root root 4.0K Sep 18 12:21 ..
drwxr-xr-x  3 root root 4.0K Sep 18 12:25 4.1.7+
drwxr-xr-x  3 root root 4.0K Sep 18 12:25 4.1.7-v7+
root@dietpi:/#

XECDesign, 2015-09-18:
Did you reboot?

Fourdee, 2015-09-18:
> Did you reboot?
Turns out I didn't lol. 

Bad day at the office, complete user error. Apologies.

closed 2015-09-18 by Fourdee

XECDesign, 2015-09-18:
Happens