Hexxeh/rpi-update #193: After Update, Raspberry Pi Crashes/Freezes

KyleBoyer, 2015-10-05:
After performing an update on my Raspberry Pi 2, and my buddies Raspberry Pi 2, we noticed that our Pis would randomly "black screen" and freeze after running for a bit. I'm not too privvy to check what logs could exist from this. The only known fix to get the pi "un-frozen" is to remove the power, and replace it. To note, the rpi-update worked successfully to update the Pi to the new version, however after this is when we started experiencing random crashing. So far, my Pi has crashed 3 times since last night, and it could be more if I was able to reset it as soon as it went down(but I couldn't because I sleep at night).

popcornmix, 2015-10-05:
Firstly are you overclocking? If so, disable that and confirm the issue still occurs.

I'm not seeing other reports of this. You can use
``sudo rpi-update <hash>``
Using the hashes (the hex number at the end of each commi's URL) from https://github.com/Hexxeh/rpi-firmware/commits/master 
If you can identify the exact update that introduced the problem, we can investigate exactly what changed.

KyleBoyer, 2015-10-05:
> Firstly are you overclocking? If so, disable that and confirm the issue still occurs.
> 
> I'm not seeing other reports of this. You can use:
> sudo rpi-update <hash>
> Using the hashes (the hex number at the end of each commi's URL) from https://github.com/Hexxeh/rpi-firmware/commits/master to downgrade to a specific firmware version.
> If you can identify the exact update that introduced the problem, we can investigate exactly what changed.

I am not overclocking. I will attempt downgrading to see if it changes anything!

KyleBoyer, 2015-10-07:
I downgraded to "608c3b3e994e086d400f77469682ff1e8f346e27" 4.1.8 and it has fixed the issue so far. Thus the problem must be happening in one of the two latest firmware revisions. After a bit more testing, I can see if 4.1.9 also does not have the issue.

popcornmix, 2015-10-07:
Thanks. I would be interested if 4.1.9 is good or bad.

KyleBoyer, 2015-10-07:
> Thanks. I would be interested if 4.1.9 is good or bad.

Just installed 4.1.9 to test.

KyleBoyer, 2015-10-07:
4.1.9 DOES have the same issue.

simoncaron, 2015-10-09:
Hi, I'm new here but I can confirm that I also have update my pi2 to the latest firmware and I'm also getting freeze. Each time messages log is about cfg80211 and CRDA. I don't know why cfg80211 module is loaded since I don't use wi-fi.

diluex, 2015-10-10:
it's the same for me. my raspberrypi is running as a server and freezing after 1-2 h and not responding over ssh after update. also reading crda an cfg80211 wihtout using wifi.

popcornmix, 2015-10-10:
Please try latest rpi-update and report back.

mbentley, 2015-10-12:
The latest has been working for me for the last 18 hours.

simoncaron, 2015-10-12:
I'll give it a try and comeback with the results.

popcornmix, 2015-10-13:
@KyleBoyer 
Pretty sure this is fixed. Please test and close if okay.

KyleBoyer, 2015-10-13:
The latest version of the OS has (so far) fixed this issue. After successfully completing 48 hours without experiencing this issue again, I believe it is safe to say that this issue can now be closed.

closed 2015-10-13 by KyleBoyer

KyleBoyer, 2015-11-30:
I experienced this again on the latest update.

popcornmix, 2015-11-30:
Can you confirm the exact firmware version that causes the problem?
Use ``sudo rpi-update <hash>`` with the has from then end of URLs from here: https://github.com/Hexxeh/rpi-firmware/commits/master

KyleBoyer, 2015-11-30:
The most recent version as of 12 hours ago.

popcornmix, 2015-11-30:
And reverting to the version before that fixes it? It's important to confirm exactly which firmware caused the problem.

KyleBoyer, 2015-11-30:
I shall try reverting when I'm back at my computer, but over this past thanksgiving(USA Holiday) week, it went down while out of town in the same manner as before(when this ticket was opened). It took a day or so to crash, so I'll keep you posted.