Hexxeh/rpi-update #196: Using /bin/sh to check for HardFP/SoftFP is unreliable

grigorig, 2015-11-09:
The /bin/sh binary may not even use any floating point instructions. So the check is unreliable and fails on Fedora 22 armhf, for instance. In turn, rpi-update will incorrectly use the SoftFP libraries.

Maybe it is more reliable to check against something that surely uses FP instructions, e.g. /lib/libm.so?

popcornmix, 2015-11-09:
Does that definitely work for you? A PR would be useful.

grigorig, 2015-11-09:
It seems to work fine, I'm still testing something. PR incoming.

grigorig, 2015-11-09:
Okay, looks like checking e.g. libm basically works, but the location of the library depends on whether you have a multilib system or not. So it's not reliably to assume a fixed path for it.

The right way to check for hardfp seems to be `readelf -h`. It outputs something like

`Flags:                             0x5000402, has entry point, Version5 EABI, hard-float ABI`

But apparently that only works for newer versions of readelf, so I'm not sure it works on Raspbian. Maybe someone can check with `readelf -h /bin/sh`?

popcornmix, 2015-11-09:
With Jessie raspbian:
```
readelf -h /bin/sh
ELF Header:
  Magic:   7f 45 4c 46 01 01 01 00 00 00 00 00 00 00 00 00 
  Class:                             ELF32
  Data:                              2's complement, little endian
  Version:                           1 (current)
  OS/ABI:                            UNIX - System V
  ABI Version:                       0
  Type:                              DYN (Shared object file)
  Machine:                           ARM
  Version:                           0x1
  Entry point address:               0x3098
  Start of program headers:          52 (bytes into file)
  Start of section headers:          102968 (bytes into file)
  Flags:                             0x5000402, has entry point, Version5 EABI, hard-float ABI
  Size of this header:               52 (bytes)
  Size of program headers:           32 (bytes)
  Number of program headers:         9
  Size of section headers:           40 (bytes)
  Number of section headers:         28
  Section header string table index: 27
```

grigorig, 2015-11-09:
I just check on Wheezy Raspbian. It doesn't work there, what a bummer. But maybe it's possible to check both ways.

popcornmix, 2015-11-09:
What does readelf output on Wheezy?
There is another answer here: http://raspberrypi.stackexchange.com/questions/4677/how-can-i-tell-if-i-am-using-the-hard-float-or-the-soft-float-version-of-debian

grigorig, 2015-11-09:
It's pretty much the same, just without the string "hard-float ABI". Supposedly the display of this was added somewhat recently. The answer in stackexchange is specific to Debian's multilib setup. It is not portable.

Maybe we can just add the readelf -h method to rpi-update in addition to the old method. What do you think?

popcornmix, 2015-11-09:
Yes, trying both schemes and if either results in hard-float then assume hard-float (99% of users will be running hard-float anyway).

Or you could decode the bit field from flags yourself.

grigorig, 2015-11-18:
I forgot to reference this issue in the PR. This was fixed by PR #199.

closed 2015-11-18 by grigorig