Hexxeh/rpi-update #197: Clearing out older firmware updates after new release installed successfully by rpi-update

shaun10, 2015-11-10:
Just wondering if anyone has started working on modifications to remove older firmware files after updates have installed successfully. In the screenshot below about 860 MB is used for older firmware versions in /lib/modules

[image]

Anyone think the option to automatically remove older versions would be helpful?

popcornmix, 2015-11-10:
Okay, it's fair point. I've added an option to delete the unused modules directories when doing an update. Try:
```sudo PRUNE_MODULES=1 rpi-update```

If you are already up to date, run
``sudo rm /boot/.firmware_revision``
first to force an update.

shaun10, 2015-11-10:
Worked perfectly:
[image]

Freed up about 20% of my 8GB card which had only 5.2GB available with NOOBs. Much more lean. 

So in future just run with PRUNE_MODULES=1 parameter to update while only keeping one of the last module dirs right?

Thanks again

popcornmix, 2015-11-10:
Yep

closed 2015-11-17 by shaun10