Hexxeh/rpi-update #200: kernel version not updating

ddonindia, 2015-12-27:
Hi,

I am trying to upgrade my kernel but for some reason uname -a still shows 3.18.0-25-rpi2.
I am booting my rpi2 from usb drive and running ubuntu 15.10.
Is it supported ?
```
ubuntu@ubuntu:/boot$ ls -rlt
total 32088
drwxr-xr-x 3 root root   16384 Jan  1  1970 firmware
lrwxrwxrwx 1 root root      19 Apr  6  2015 config.txt -> firmware/config.txt
lrwxrwxrwx 1 root root      20 Apr  6  2015 cmdline.txt -> firmware/cmdline.txt
-rw------- 1 root root 3337648 Jul  5 17:10 vmlinuz-3.18.0-25-rpi2
-rw-r--r-- 1 root root  119705 Jul  5 17:10 config-3.18.0-25-rpi2
-rw-r--r-- 1 root root  685516 Jul  5 17:10 abi-3.18.0-25-rpi2
-rw------- 1 root root 2346158 Jul  5 17:10 System.map-3.18.0-25-rpi2
-rw-r--r-- 1 root root 6205013 Nov  4 02:35 initrd.img-3.18.0-25-rpi2
drwxr-xr-x 2 root root    4096 Dec 27 15:20 overlays
-rw-r--r-- 1 root root 2731832 Dec 27 15:33 start.elf
-rw-r--r-- 1 root root  605272 Dec 27 15:33 start_cd.elf
-rw-r--r-- 1 root root 4878728 Dec 27 15:33 start_db.elf
-rw-r--r-- 1 root root 3831848 Dec 27 15:33 start_x.elf
-rw-r--r-- 1 root root   17900 Dec 27 15:33 bootcode.bin
-rw-r--r-- 1 root root    9696 Dec 27 15:33 fixup_x.dat
-rw-r--r-- 1 root root    9692 Dec 27 15:33 fixup_db.dat
-rw-r--r-- 1 root root    2472 Dec 27 15:33 fixup_cd.dat
-rw-r--r-- 1 root root    6450 Dec 27 15:33 fixup.dat
-rw-r--r-- 1 root root 3946040 Dec 27 15:33 kernel.img
-rw-r--r-- 1 root root 4024732 Dec 27 15:33 kernel7.img
-rw-r--r-- 1 root root   11479 Dec 27 15:33 bcm2709-rpi-2-b.dtb
-rw-r--r-- 1 root root   10242 Dec 27 15:33 bcm2708-rpi-cm.dtb
-rw-r--r-- 1 root root   10212 Dec 27 15:33 bcm2708-rpi-b.dtb
-rw-r--r-- 1 root root   10491 Dec 27 15:33 bcm2708-rpi-b-plus.dtb
ubuntu@ubuntu:/boot$ uname -a
Linux ubuntu 3.18.0-25-rpi2 #26-Ubuntu SMP PREEMPT Sun Jul 5 06:46:34 UTC 2015 armv7l armv7l armv7l GNU/Linux
ubuntu@ubuntu:/boot$ cat cmdline.txt
dwc_otg.lpm_enable=0 console=tty1 root=/dev/sda2 rootfstype=ext4 elevator=deadline  rootwait
#dwc_otg.lpm_enable=0 console=tty1 root=/dev/mmcblk0p2 rootfstype=ext4 elevator=deadline rootwait

Distributor ID: Ubuntu
Description:    Ubuntu 15.10
Release:        15.10
Codename:       wily

```

popcornmix, 2015-12-27:
rpi-upate is designed for raspbian. Other distributions use custom kernels and have their own mechanisms for upgrading (presumably through apt-get with Ubuntu).
Using a raspbian kernel with Ubuntu may well break features that Ubuntu expects from its kernel.
Use raspbian is you need a newer kernel, or ask in an Ubuntu forum for advice on upgrading an Ubuntu kernel.

closed 2015-12-27 by popcornmix

ddonindia, 2015-12-27:
Thanks. Sorry I was not aware rpi-upate is only for raspbian.