Hexxeh/rpi-update #203: Current version

vlmxs, 2016-01-22:
How can I check the current stable version? 

I am on: "Linux raspberrypi 4.1.15+ #830 Tue Dec 15 16:58:28 GMT 2015 armv6l GNU/Linux
pi@raspberrypi:~ $" but which one will be installed?

Sorry if "noob" question :)

popcornmix, 2016-01-22:
There isn't a stable version from rpi-update.
If you want a stable version don't use rpi-update.
It is designed purely for bleeding edge testers and users suffering from a bug that has only recently been fixed.
``apt-get upgrade`` will get you the version recommended.

vlmxs, 2016-01-31:
Ok, I was not very specific. So I have certain version of the firmware. How can I check if this is "old" or relatively "new" version in relation to current version available through "rpi-update" utility? Where I can find a "changelog" or "version" history?

vlmxs, 2016-01-31:
When I type "sudo rpi-update" I do not know to which version the firmware will upgrade, if there is no newest version available it has to stop, saying that the newest firmware is already installed, but I think it just overwrites the existing firmware although they might be the same.

popcornmix, 2016-02-01:
If your firmware is up to date, then you will see ``*** Your firmware is already up to date`` and rpi-update will not touch the firmware.
If you want to see if there is an update without updating, then:
``sudo JUST_CHECK=1 ri--update``

vlmxs, 2016-02-05:
Thank you for the very detailed answer. It helped me!

closed 2016-02-06 by popcornmix