Hexxeh/rpi-update #204: Unable to get local issuer certificate

timtlm, 2016-03-23:
I've tried for the life of me to get this to run using multiple different versions of commands, but ultimately I'm getting an Unable to get local issuer certificate error.  I'm running raspbian (retropie) and I've been able to successfully install rpi-update using:
`sudo apt-get install rpi-update`
Running the update after that I get the issuer certificate error (I believe during the self update step).  I've run the troubleshooting commands to make sure ca-certificates is installed, and that the time is set.  I've also run it using:
`sudo UPDATE_SELF=0 rpi-update`
Then I get a message of "No hash received from github: https://api.github.com/repos/Hexxeh/rpi-firmware/git/refs/heads/master" followed by the above mentioned certificate error.

popcornmix, 2016-03-23:
Can you run this command and report the output:
```curl https://raw.githubusercontent.com/Hexxeh/rpi-update/master/rpi-update -o /tmp/rpi-update```

timtlm, 2016-03-23:
Thanks for the reply.  I ended up putting a fresh image of raspbian lite on another sd card, and I was able to run rpi-update successfully.  I'm not sure what the issue is with the retropie image that prevents this from running successfully.  If I run any curl command on the retropie image I get the certificate error.  I suppose I could try purging curl and reinstalling it.  However, since I got the firmware updated, I think you can just close this.

popcornmix, 2016-03-23:
Yes, sounds like retropie image issue if all https accesses fail.

closed 2016-03-23 by popcornmix