Hexxeh/rpi-update #206: Failed to down update for rpi-update

stayupthetree, 2016-05-15:
' !!! Failed to download update for rpi-update!
 !!! Make sure you have ca-certificates installed and that the time is set correctly

I have read the readme. I made sure ca-certificates was installed, even did a dpkg reconfigure.
Made sure time was correct with the ntpupdate.

Tried alternative steps cloning the repo. Same SSL Cert error.

-k and --insecure did nothing to prevent it

stayupthetree, 2016-05-15:
I was able to make some progress by editing rpi-update and adding -k to every curl command. Then I realized when I ran rpi-update it overwrote my changes.

I then discovered this gem 'sudo UPDATE_SELF=0 ./rpi-update'

Seems to be running fine currently.

Ruffio, 2016-05-20:
@stayupthetree I just ran the rpi-update without issues on a clean Raspbian image without installing any certificates or dpkg reconfigure. What is your setup? Could you try with a clean image?

sanelson2000, 2016-05-28:
I'm having the same issue, on a Raspberry Pi 3 with Jessie kernel version 4.4.11-v7+.  I'm using the RetroPie image.  I noticed that rpi-update wasn't even installed, installed it myself, and found it not working.

Ruffio, 2016-07-03:
I just installed Rasbian and Lite and har no problems  at all. Could you check the time and CA certificates?

rillian, 2016-07-05:
I think this is an issue with the `curl` package on the retropie and/or rasbian jessie images. I have the same problem, and a generic `curl --head https://github.com/` fails with a certificate validation error.

Meanwhile `curl --cacert /etc/ssl/certs/ca-certificates.crt --head https://github.com/` succeeds, so there's something wrong with the default curl path.

Using the corresponding environment variable I can work around the issue
```sh
sudo CURL_CA_BUNDLE=/etc/ssl/certs/ca-certificates.crt rpi-update
```
```

popcornmix, 2016-07-06:
@rillian can you confirm that you have this issue with a clean raspbian jessie image?
It certainly works fine for me.

Ruffio, 2016-08-01:
@rillian is this still an issue? Have you tried with a clean Raspbian Jessie image? I can't recreate your issue.

KevinBEvv, 2016-10-03:
I found that I was having trouble with DNS, which was caused by a firewall restriction (not apart of the RaspPi Raspbian install, but a separate issue).  This caused the error for me.

Ruffio, 2017-01-22:
@stayupthetree is this still an issue? If not, please close it. Thanks.

seyyah, 2017-04-06:
I have this issue with my ~~clean~~ raspbian jessie image.

```sh
$ sudo apt-get update
$ sudo apt-get upgrade

$ lsb_release -a
No LSB modules are available.
Distributor ID: Raspbian
Description:    Raspbian GNU/Linux 8.0 (jessie)
Release:        8.0
Codename:       jessie

$ uname -a
Linux raspberrypi 4.4.50-v7+ #970 SMP Mon Feb 20 19:18:29 GMT 2017 armv7l GNU/Linux

$ sudo rpi-update -k
 *** Raspberry Pi firmware updater by Hexxeh, enhanced by AndrewS and Dom
 *** Performing self-update
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0curl: (60) SSL certificate problem: unable to get local issuer certificate
More details here: http://curl.haxx.se/docs/sslcerts.html

 of Certificate Authority (CA) public keys (CA certs). If the default
 bundle file isn't adequate, you can specify an alternate file
 using the --cacert option.
If this HTTPS server uses a certificate signed by a CA represented in
 the bundle, the certificate verification probably failed due to a
 problem with the certificate (it might be expired, or the name might
 not match the domain name in the URL).
If you'd like to turn off curl's verification of the certificate, use
 the -k (or --insecure) option.
 !!! Failed to download update for rpi-update!
 !!! Make sure you have ca-certificates installed and that the time is set correctly

$ sudo CURL_CA_BUNDLE=/etc/ssl/certs/ca-certificates.crt rpi-update
 *** Raspberry Pi firmware updater by Hexxeh, enhanced by AndrewS and Dom
 *** Performing self-update
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100 12762  100 12762    0     0  20237      0 --:--:-- --:--:-- --:--:-- 20225
 *** Relaunching after update
 *** Raspberry Pi firmware updater by Hexxeh, enhanced by AndrewS and Dom
 *** We're running for the first time
 *** Backing up files (this will take a few minutes)
 *** Remove old firmware backup
 *** Backing up firmware
 *** Remove old modules backup
 *** Backing up modules 4.4.50-v7+
#############################################################
WARNING: This update bumps to rpi-4.9.y linux tree
Be aware there could be compatibility issues with some drivers
Discussion here:
https://www.raspberrypi.org/forums/viewtopic.php?f=29&t=167934
##############################################################
Would you like to proceed? (y/N)
 *** Downloading specific firmware revision (this will take a few minutes)
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100   168    0   168    0     0    232      0 --:--:-- --:--:-- --:--:--   232
100 53.7M  100 53.7M    0     0   507k      0  0:01:48  0:01:48 --:--:--  866k
 *** Updating firmware
 *** Updating kernel modules
 *** depmod 4.9.20+
 *** depmod 4.9.20-v7+
 *** Updating VideoCore libraries
 *** Using HardFP libraries
 *** Updating SDK
 *** Running ldconfig
 *** Storing current firmware revision
 *** Deleting downloaded files
 *** Syncing changes to disk
 *** If no errors appeared, your firmware was successfully updated to 3ff94f1fc459f88d3e2530542fb609643a7bd1a6
 *** A reboot is needed to activate the new firmware
```

popcornmix, 2017-04-06:
@seyyah does ``date`` report the correct date/time?

seyyah, 2017-04-06:
I configure date with `dpkg-reconfigure tzdata # Asia -> Istanbul` before `sudo rpi-update` OR `sudo rpi-update -k`. So,

```sh
$ sudo dpkg-reconfigure tzdata

$ sudo rpi-update
...
!!! Make sure you have ca-certificates installed and that the time is set correctly

$ sudo rpi-update -k
...
!!! Make sure you have ca-certificates installed and that the time is set correctly

$ sudo CURL_CA_BUNDLE=/etc/ssl/certs/ca-certificates.crt rpi-update
...
 *** A reboot is needed to activate the new firmware
```

popcornmix, 2017-04-06:
But does ``date`` report the correct date and time?

seyyah, 2017-04-06:
Yes, `date` report the correct/locale date and time,

```sh
$ LC_ALL=C date -u; \
LC_ALL=C TZ="America/Los_Angeles" date; \
LC_ALL=C TZ="Asia/Istanbul" date; \
LC_ALL=C date
Thu Apr  6 14:53:23 UTC 2017
Thu Apr  6 07:53:23 PDT 2017
Thu Apr  6 17:53:23 +03 2017 <-- local date and time and it is correct
Thu Apr  6 17:53:23 +03 2017
```

popcornmix, 2017-04-06:
If you run
```
curl -Lfs --output /tmp/rpi-update https://raw.githubusercontent.com/Hexxeh/rpi-update/master/rpi-update
```
do you get the same error?

seyyah, 2017-04-06:
I will try and report as soon as possible.

seyyah, 2017-04-11:
I get below the same error,

```sh
$ curl -Lfs --output /tmp/rpi-update https://raw.githubusercontent.com/Hexxeh/rpi-update/master/rpi-update

<not write anything on the screen>

$ cat /tmp/rpi-update
cat: /tmp/rpi-update: No such file or directory

$ curl https://raw.githubusercontent.com/Hexxeh/rpi-update/master/rpi-update
curl: (60) SSL certificate problem: unable to get local issuer certificate
More details here: http://curl.haxx.se/docs/sslcerts.html

curl performs SSL certificate verification by default, using a "bundle"
 of Certificate Authority (CA) public keys (CA certs). If the default
 bundle file isn't adequate, you can specify an alternate file
 using the --cacert option.
If this HTTPS server uses a certificate signed by a CA represented in
 the bundle, the certificate verification probably failed due to a
 problem with the certificate (it might be expired, or the name might
 not match the domain name in the URL).
If you'd like to turn off curl's verification of the certificate, use
 the -k (or --insecure) option.

$ curl -k https://raw.githubusercontent.com/Hexxeh/rpi-update/master/rpi-update
#!/bin/bash

set -o nounset
set -o errexit

REPO_URI=${REPO_URI:-"https://github.com/Hexxeh/rpi-firmware"}
...
```

Ruffio, 2017-05-16:
@stayupthetree can this issue be closed?

hgovindarajan, 2017-08-12:
I ran into this issue when I attempted to setup & update my rasp-pi prior to configuring my wifi adapter. Doh!

popcornmix, 2017-08-13:
No response from OP so closing. Feel free to reopen if there is new information.

closed 2017-08-13 by popcornmix

amrithmmh, 2018-03-26:
i am getting this error now after updating and upgrading to stretch

zimtzeke, 2018-10-24:
for me this worked fine:
$ sudo CURL_CA_BUNDLE=/etc/ssl/certs/ca-certificates.crt rpi-update

orjanv, 2019-03-21:
Getting this error in jessie, tried the above with no luck.

simonliu009, 2020-03-18:
I accidently  typed "N" after `Would you like to proceed? (y/N)` and I got the same error when I tried to run `sudo rpi-update`again.  `date` command returned the correct time. I tried removing and reinstalling `ca-certificates` and no luck.  Then I walked out for around 40 minutes and tried again when I came back , everything works fine now.

zhangyoufu, 2020-11-07:
FYI, `raw.githubusercontent.com` is blocked in mainland China.

`sudo UPDATE_SELF=0 rpi-update` to skip self update or edit `/etc/hosts` manually to workaround DNS pollution.

vizshrc, 2020-11-22:
> FYI, `raw.githubusercontent.com` is blocked in mainland China.
> 
> `sudo UPDATE_SELF=0 rpi-update` to skip self update or edit `/etc/hosts` manually to workaround DNS pollution.

Right

arvinxian, 2021-01-19:
> FYI, `raw.githubusercontent.com` is blocked in mainland China.
> 
> `sudo UPDATE_SELF=0 rpi-update` to skip self update or edit `/etc/hosts` manually to workaround DNS pollution.

it works, in mainland China

popcornmix, 2022-07-07:
@geerlingguy if you are currently seeing this issue, then report output of:
```
date
curl -Lfs --output /tmp/rpi-update https://raw.githubusercontent.com/Hexxeh/rpi-update/master/rpi-update
```

652436962, 2022-07-12:
> FYI, `raw.githubusercontent.com` is blocked in mainland China.
> 
> `sudo UPDATE_SELF=0 rpi-update` to skip self update or edit `/etc/hosts` manually to workaround DNS pollution.

can you provide your email  ?   I  can't come true

652436962, 2022-07-13:
> > 仅供参考，`raw.githubusercontent.com`在中国大陆被封锁。
> > `sudo UPDATE_SELF=0 rpi-update`跳过自我更新或`/etc/hosts`手动编辑以解决 DNS 污染问题。
> 
> 有效，在中国大陆

are  you update  sucess ?   what  you  do?   can you provide  your email?

hegdenischay, 2023-09-02:
FYI for the Indians looking at this issue, the ISP Jio blocks `raw.githubusercontent.com` for some reason. Changing the DNS to anything other than the ISP's DNS fixes this issue.

krishnaTORQUE, 2023-09-02:
> 

Might not true.
Because I am using airtel fiber.

hegdenischay, 2023-09-02:
@krishnaTORQUE This is not an India-wide problem, this is specifically a Reliance Jio problem. They're far too trigger happy in blocking websites, and I haven't had this problem elsewhere.

krishnaTORQUE, 2023-09-02:
> @krishnaTORQUE This is not an India-wide problem, this is specifically a Reliance Jio problem. They're far too trigger happy in blocking websites, and I haven't had this problem on other ISPs.

Yes I got it.
But I am facing the same issue but I am using airtel fiber.
If jio the only provider blocking then I should not face this issue but I do face.