Hexxeh/rpi-update #209: New kernel not used after install

Izin, 2016-06-25:
Hi,

I used the command `rpi-update` to update the kernel. Now I have the kernel 4.4.13 "installed"

But after 2 reboot, the kernel used by Raspbian is still the 3.4 (unarme -r).

When I display the content of the folder `/lib/modules/`, I see this:
```
malo@tiger ~ $ ls -la /lib/modules/
total 28
drwxr-xr-x  7 root root 4096 Jun 25 21:48 .
drwxr-xr-x 13 root root 4096 Jun 25 21:33 ..
drwxrwxr-x  3 1000 pi   4096 Dec 23  2014 3.4.103
drwxr-xr-x  3 root root 4096 Jun 25 21:48 4.4.13+
drwxr-xr-x  3 root root 4096 Jun 25 21:48 4.4.13-v7+
```

Do I have to delete the `3.4.103` folder, or change the user/group?

Did I misunderstood something?

Thanks!

popcornmix, 2016-06-25:
Can you report the output of
```
uname -a
vcgencmd version
```
Are you using NOOBS? Do you believe it is Raspbian Wheezy or Jessie or is it some unofficial image?

Izin, 2016-06-25:
Wow, you answered me so quickly!

```
Linux tiger 3.4.103 #1 SMP PREEMPT Thu Dec 18 13:07:12 CST 2014 armv7l GNU/Linux
```
```
-bash: vcgencmd: command not found
```

I'm using the "
Raspbian For BananaPi v1412" image downloaded [from here](http://forum.lemaker.org/portal.php?mod=view&aid=29).

It's a Wheezy Raspbian. I think it's kinda unoficial by the way.

popcornmix, 2016-06-25:
If you are using a banana pi then I can't help.
rpi-update only makes sense on a raspberry pi.
Banana Pi will have different kernel and firmware.

closed 2016-06-25 by popcornmix

Izin, 2016-06-26:
I understand... Thank you for your answers btw :+1: