Hexxeh/rpi-update #210: Running from a bash script brakes down

szmaorka, 2016-06-26:
Running from shell goes well. But when I want to run it from my `update.sh` bash script, it gives me an error:
```
gzip: stdin: not in gzip format
tar: Child returned status 1
tar: Error is not recoverable: exiting now
```
Thanks! :)

lurch, 2016-06-30:
The *only* (optional) parameter that rpi-update accepts is the git hash of a specific revision to update to https://github.com/Hexxeh/rpi-update#options

Unsurprisingly, there's no firmware revision that has a git hash of "-y"  ;-)

(although maybe there could be a check that the supplied hash is actually a valid commit ID, before piping the output of curl directly into tar)

Ruffio, 2016-07-24:
@szmaorka can this issue be closed?

szmaorka, 2016-07-24:
@Ruffio Have been added a warning if using an invalid parameter? If yes, of course we can close :)

Ruffio, 2016-07-24:
@lurch ?

popcornmix, 2016-07-24:
See: https://github.com/Hexxeh/rpi-update/commit/b19779b2c8b769b3bd2bc9fb4cf858f340a11cf5

@szmaorka @lurch please test.

szmaorka, 2016-07-24:
@popcornmix Tested, works. Gives `Invalid git hash specified` error!

closed 2016-07-24 by szmaorka