Hexxeh/rpi-update #212: curl: (35) Unknown SSL protocol error in connection to github.com:443

rezna11, 2016-07-13:
Won't run for me at :( Always get this error. Tried downgrade curl to 7.3.3. No change

```
pi@raspberrypi:~ $ sudo rpi-update 33a6707cf1c96b8a2b5dac2ac9dead590db9fcaa
 *** Raspberry Pi firmware updater by Hexxeh, enhanced by AndrewS and Dom
 *** Performing self-update
 *** Relaunching after update
 *** Raspberry Pi firmware updater by Hexxeh, enhanced by AndrewS and Dom
 *** We're running for the first time
 *** Backing up files (this will take a few minutes)
 *** Remove old firmware backup
 *** Backing up firmware
 *** Remove old modules backup
 *** Backing up modules 4.4.11-v7+
 *** Downloading specific firmware revision (this will take a few minutes)
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0     0    0     0    0     0      0      0 --:--:--  0:02:59 --:--:--     0curl: (35) Unknown SSL protocol error in connection to github.com:443

gzip: stdin: unexpected end of file
tar: Child returned status 1
tar: Error is not recoverable: exiting now
```

popcornmix, 2016-07-13:
Are you using raspbian jessie?
what do the following report?
```
vcgencmd version
uname -a
```
Does this work for you?
```
wget -L https://codeload.github.com/Hexxeh/rpi-firmware/legacy.tar.gz/33a6707cf1c96b8a2b5dac2ac9dead590db9fcaa
```

rezna11, 2016-07-13:
I'm on jessie, yes and wget fails too:

```
pi@raspberrypi:~ $ vcgencmd version
May 20 2016 19:01:31
Copyright (c) 2012 Broadcom
version faf071dd4885c5ac1a89483d35a5326e7f69495f (clean) (release)
pi@raspberrypi:~ $ uname -a
Linux raspberrypi 4.4.11-v7+ #888 SMP Mon May 23 20:10:33 BST 2016 armv7l GNU/Linux
pi@raspberrypi:~ $ wget -L https://codeload.github.com/Hexxeh/rpi-firmware/legacy.tar.gz/33a6707cf1c96b8a2b5dac2ac9dead590db9fcaa
--2016-07-13 11:40:11--  https://codeload.github.com/Hexxeh/rpi-firmware/legacy.tar.gz/33a6707cf1c96b8a2b5dac2ac9dead590db9fcaa
Resolving codeload.github.com (codeload.github.com)... 192.30.253.121
Connecting to codeload.github.com (codeload.github.com)|192.30.253.121|:443... connected.

```

just stuck on the last line

popcornmix, 2016-07-13:
Are you using ethernet or wifi? It looks like you have internet connectivity issues.
Can you download the same file from another site?
```wget -L https://dl.dropboxusercontent.com/u/3669512/temp/33a6707cf1c96b8a2b5dac2ac9dead590db9fcaa```

rezna11, 2016-07-13:
wired internet

downloading from the other location works fine:

```
pi@raspberrypi:~ $ wget -L https://dl.dropboxusercontent.com/u/3669512/temp/33a6707cf1c96b8a2b5dac2ac9dead590db9fcaa
--2016-07-13 14:24:35--  https://dl.dropboxusercontent.com/u/3669512/temp/33a6707cf1c96b8a2b5dac2ac9dead590db9fcaa
Resolving dl.dropboxusercontent.com (dl.dropboxusercontent.com)... 108.160.173.165
Connecting to dl.dropboxusercontent.com (dl.dropboxusercontent.com)|108.160.173.165|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 48528681 (46M) [application/octet-stream]
Saving to: ‘33a6707cf1c96b8a2b5dac2ac9dead590db9fcaa’

a6707cf1c96b8a2b5dac2ac9dead590db   1%[                                                             ] 551.23K  51.8KB/s   eta 19m 50s
```

popcornmix, 2016-07-13:
Can you run the wget command on another (non-pi) machine in your house?

rezna11, 2016-07-13:
thanks for pointing in the right direction, it was a network issue...
after disabling EOIP tunnel on the router, it now works... why I had to do that, no idea, especially when other https downloads worked

Ruffio, 2016-07-24:
@rezna11 as the issue has been resolved, please close this issue. Thanks

closed 2016-07-25 by popcornmix