Hexxeh/rpi-update #215: current firmware and update available

C4Wiz, 2016-08-11:
is it possible to have rpi-update tell you what the current version installed is and what you are upgrading to?

popcornmix, 2016-08-11:
Try:
```sudo JUST_CHECK=1 rpi-update```

C4Wiz, 2016-08-11:
$ sudo JUST_CHECK=1 rpi-update
 *** Raspberry Pi firmware updater by Hexxeh, enhanced by AndrewS and Dom
 *** Performing self-update
 *** Relaunching after update
 *** Raspberry Pi firmware updater by Hexxeh, enhanced by AndrewS and Dom
 *** Your firmware is already up to date

popcornmix, 2016-08-12:
Try it again now as there has been a new update.

C4Wiz, 2016-08-15:
nope still no notification of what you have or what is being downloaded, it just self updates and starts to download new firmware

popcornmix, 2016-08-15:
It won't download new firmware with ``JUST_CHECK=1``.
Can you report exactly what you typed in and what the output was?

C4Wiz, 2016-08-15:
it is posted above

popcornmix, 2016-08-15:
But that looks like you ran:
``sudo rpi-update``
You are supported to run:
``sudo JUST_CHECK=1 rpi-update``
to see what (if anything) has changed before updating.

C4Wiz, 2016-08-15:
using JUST_CHECK=1

 sudo JUST_CHECK=1 rpi-update
 *** Raspberry Pi firmware updater by Hexxeh, enhanced by AndrewS and Dom
 *** Performing self-update
 *** Relaunching after update
 *** Raspberry Pi firmware updater by Hexxeh, enhanced by AndrewS and Dom
 *** Your firmware is already up to date

still 0 info on current

C4Wiz, 2016-08-15:
it should say by default!!

popcornmix, 2016-08-15:
If you want to see what you get when there are changes then you can fake it:
```
pi@domnfs:~ $ echo 39054eaf0d3cfa351f2e23126e0ae88596a9437d | sudo tee /boot/.firmware_revision 
39054eaf0d3cfa351f2e23126e0ae88596a9437d
pi@domnfs:~ $ sudo JUST_CHECK=1 rpi-update
 *** Raspberry Pi firmware updater by Hexxeh, enhanced by AndrewS and Dom
 *** Performing self-update
 *** Relaunching after update
 *** Raspberry Pi firmware updater by Hexxeh, enhanced by AndrewS and Dom
 *** Firmware update required. New commits available:
======================================================
Commit: firmware: Pushed so files missed from last update
======================================================
Commit: firmware: MJPEG enc: Don't hold JPEG module the whole time\nSee: https://www.raspberrypi.org/forums/viewtopic.php?f=70&t=153445
Commit: firmware: image_encode: Add parameter to set the restart interval\nSee: https://www.raspberrypi.org/forums/viewtopic.php?f=38&t=153553
Commit: firmware: Raspistill: Add option to set the JPEG restart interval\nSee: https://www.raspberrypi.org/forums/viewtopic.php?f=38&t=153553
Commit: firmware: rawcam: Set min line interrupt rate to 16\nSee: https://www.raspberrypi.org/forums/viewtopic.php?f=43&t=47798&start=75#p1006778
Commit: firmware: image_fx: Fix some issues with frameheight/sliceheight\nfirmware: image_fx: Fix some YUVUV issues with pitch and vpitch\nfirmware: deinterlace: Allow interlaced state of video buffer to be signalled from host
======================================================
Commit: kernel: Bump to 4.4.15
Commit: kernel: BCM270X_DT: Declare the audio PWM pins explicitly\nSee: https://github.com/raspberrypi/linux/issues/1473
Commit: kernel: Add support for Dion Audio LOCO DAC-AMP HAT\nSee: https://github.com/raspberrypi/linux/pull/1553
Commit: kernel: BCM270X_DT: Overlay to re-enable HW CS on SPI0\nSee: https://github.com/raspberrypi/linux/issues/1547
======================================================
Commit: firmware: video_render: Ensure image structure is initialised
Commit: firmware: image_fx: Add assert to fatal error code path
======================================================
Commit: firmware: arm_dt: allow audio output without dtparam=audio=on\nSee: https://www.raspberrypi.org/forums/viewtopic.php?f=66&t=154604
Commit: firmware: dt-blob: Add BT_ON and WL_ON pins, and use if present
======================================================
Commit: kernel: Bump to 4.4.16
Commit: kernel: bcm2708_fb: Default framebuffer to 32-bit
Commit: firmware: platform: Default to 16bit framebuffer with cutdown firmware
Commit: firmware: armstub: Move redundant write to mailbox register\nSee: https://github.com/raspberrypi/firmware/issues/578
Commit: firmware: vc_client: Fix for sending vguPolygon data in chunks\nSee: https://github.com/raspberrypi/firmware/issues/633
======================================================
Commit: kernel: fix auto-sense in lirc_rpi driver\nSee: https://github.com/raspberrypi/linux/pull/1581
Commit: kernel: enable gembird joypad support\nSee: https://github.com/raspberrypi/linux/pull/1589
Commit: kernel: Added HiFiBerry Digi+ Pro driver\nSee: https://github.com/raspberrypi/linux/pull/1583
Commit: kernel: overlays: Add assert_falling_edge to pps-gpio overlay\nSee: https://github.com/raspberrypi/linux/issues/1590
Commit: kernel: Experimental graphics driver DSI support\nSee: https://github.com/raspberrypi/linux/pull/1556
Commit: firmware: Add customisation project board type\nfirmware: platform: Enable Bluetooth on custom
Commit: firmware: gencmd: Add support for hdmi_cvt and hdmi_timings\nSee: https://github.com/raspberrypi/firmware/issues/637
Commit: firmware: arm_display: Allow source aspect ratio to be configured\nSee: https://github.com/raspberrypi/firmware/issues/638
Commit: firmware: mmal: Allow /dev/vchiq to be opened initialised prior to mmal_vc_init
======================================================
Commit: kernel: Bump to 4.4.17
Commit: kernel: config: Enable SENSORS_LM75\nSee: https://github.com/raspberrypi/firmware/issues/508
Commit: kernel: config: Enable SERIAL_SC16IS7XX\nSee: https://github.com/raspberrypi/linux/issues/1594
Commit: kernel: snd-bcm2835: Don't allow responses from VC to be interrupted by user signals\nSee: https://github.com/raspberrypi/linux/issues/1560
```

C4Wiz, 2016-08-15:
just want a simple, current version is 4.4.15, available version 4.4.17, would you like to update?y/n?

popcornmix, 2016-08-17:
That doesn't make sense as most updates don't affect the kernel version.
What happens if there were just gpu changes since last update?

C4Wiz, 2016-08-17:
then those would be displayed as above

C4Wiz, 2016-08-17:
right now there is 0 info on what is being updated, it just updates

popcornmix, 2016-08-17:
It is up to you. Run:
``sudo JUST_CHECK=1 rpi-update``
to check what has changed. If you want the changes run
``sudo rpi-update``

You may also check the github repo for commits (and can even follow it to get email notifications).
You could always create a trivial bash script to run the two commands with a prompt in between.
You can use [read](http://stackoverflow.com/questions/1885525/how-do-i-prompt-a-user-for-confirmation-in-bash-script) for this.

C4Wiz, 2016-08-17:
right, but it should be built into rpi-update by default is what im saying, why run 2 commands?

donaldbraman, 2016-08-23:
@C4Wiz There is an interesting discussion to be had about defaults, but probably won't be prompted by suggesting that adding all the information and a prompt is obviously the right solution.  You'd probably get more traction by thinking through the various tradeoffs, looking at other update schemes that provide more data and prompts, and laying the pros and cons out in a separate thread.  A this point, you can do everything you want with two lines of very simple code.  It might make sense to reverse the default, but that should also allow people to silence the information and skip the prompts if they like.  At that point, it's a broader design question.  You may be right, just sayin.

WayneKeenan, 2017-03-03:
To get the git commit hash of a rpi-update run:

`cat /boot/.firmware_revision 
`
You will see something like:
> 2ca627126e49c152beb1bf7abd7122ce076dcc65

You can then go look at the commit message rpi-update firmware repo to get a clue, e.g.:

`https://github.com/Hexxeh/rpi-firmware/commit/2ca627126e49c152beb1bf7abd7122ce076dcc65
`
And see something like 'kernel bump to x.y.z' 

On a freshly built SDCard image then /boot/.firmware_revision doesn't exist because rpi-update hasn't been run.  (I'm guessing)

Some questions in my mind pop up as I don't know the release process enough:

1. Is the firmware in a Raspbian release (directly) taken from the rpi-update firmware repo?  If so... 
2. Is tagging the commit in the firmware repo with the Raspbian release (date?)doable
3. Is populating .firmware_revision on the .img doable?

Thanks
Wayne

raugusto1, 2017-05-09:
pi@ron_pi:~ $ sudo rpi-update 
 *** Raspberry Pi firmware updater by Hexxeh, enhanced by AndrewS and Dom
 *** Performing self-update
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100 12762  100 12762    0     0  35801      0 --:--:-- --:--:-- --:--:-- 35747
 *** Relaunching after update
 *** Raspberry Pi firmware updater by Hexxeh, enhanced by AndrewS and Dom
 *** We're running for the first time
 *** Backing up files (this will take a few minutes)
 *** Backing up firmware
 *** Backing up modules 4.4.50-v7+
#############################################################
This update bumps to rpi-4.9.y linux tree
Be aware there could be compatibility issues with some drivers
Discussion here:
https://www.raspberrypi.org/forums/viewtopic.php?f=29&t=167934
##############################################################
 *** Downloading specific firmware revision (this will take a few minutes)
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100   168    0   168    0     0    608      0 --:--:-- --:--:-- --:--:--   610
100 53.8M  100 53.8M    0     0   813k      0  0:01:07  0:01:07 --:--:--  293k
 *** Updating firmware
 *** Updating kernel modules
 *** depmod 4.9.26-v7+
 *** depmod 4.9.26+
 *** Updating VideoCore libraries
 *** Using HardFP libraries
 *** Updating SDK
Segmentation fault (core dumped)

 *** Running ldconfig
 *** Storing current firmware revision
 *** Deleting downloaded files
 *** Syncing changes to disk
 *** If no errors appeared, your firmware was successfully updated to b3df66512cfecc82aa3da850879f4eb3ed94deef
 *** A reboot is needed to activate the new firmware
pi@ron_pi:~ $ 
pi@ron_pi:~ $

popcornmix, 2017-05-09:
@raugusto1 I'm not sure how your problem relates to this github issue.
I assume you are referring to:
> Segmentation fault (core dumped)

That didn't come from rpi-update which completed happily.
Updating the /opt/vc/lib libraries and running ldconfig can cause applications that are using the libs to crash (I seem this with kodi) which is where the message comes from.
In general it is advisable to close any running programs that may be using GPU libs before running rpi-update, but apart from the possibly of the application crashing, it is harmless. Reboot and you should find kernel/firmware have been updated successfully.

Ruffio, 2017-05-16:
@C4Wiz has this issue been resolved?

C4Wiz, 2017-05-16:
no, still does not show any info when running sudo rpi-update.

i don't want to have to run different commands to check then update.

when sudo rpi-update is run the first time it tells you the current version installed and the version you are about to install, why can't you just make it do this EVERYTIME it is ran?

MichaIng, 2018-01-10:
I agree, was also never happy with this `bumps to rpi-4.9.y` info, that is worth nothing.
I agree that current and update kernel versions (even if they are identical, also good to know) + changelog should be always displayed, if any update is available/running. As updating takes some time, this would give us the possibility to study the changes while update is running.

I know it sounds little bit funny that I already agree to upgrade before reading the changelog, but I never had problems with them, so trust and apply every update available. It is just some interesting lecture/info to have the changes displayed besides.

By default e.g. show changes and then ask if update is wanted, would be perhaps a more logic way. But it would break all scripts, that assume the update to run automatically, thus this behaviour should be kept. Users learned that, if they want to have the decision, they have to use `JUST_CHECK=1` or study github for changes. So keep as it is, just for cosmetic/informational reasons add changelog + kernel versions.

Ruffio, 2021-03-19:
Can this be closed?