Hexxeh/rpi-update #216: raspberry pi 3 only reports 117MB ram on next

drcrallen, 2016-08-17:
The following `/boot/config.txt` only has 117MB ram when using `next`

```
# For more options and information see
# http://www.raspberrypi.org/documentation/configuration/config-txt.md
# Some settings may impact device functionality. See link above for details

# uncomment if you get no picture on HDMI for a default "safe" mode
#hdmi_safe=1

# uncomment this if your display has a black border of unused pixels visible
# and your display can output without overscan
#disable_overscan=1

# uncomment the following to adjust overscan. Use positive numbers if console
# goes off screen, and negative if there is too much border
#overscan_left=16
#overscan_right=16
#overscan_top=16
#overscan_bottom=16

# uncomment to force a console size. By default it will be display's size minus
# overscan.
#framebuffer_width=1280
#framebuffer_height=720

# uncomment if hdmi display is not detected and composite is being output
#hdmi_force_hotplug=1

# uncomment to force a specific HDMI mode (this will force VGA)
#hdmi_group=1
#hdmi_mode=1

# uncomment to force a HDMI mode rather than DVI. This can make audio work in
# DMT (computer monitor) modes
#hdmi_drive=2

# uncomment to increase signal to HDMI, if you have interference, blanking, or
# no display
#config_hdmi_boost=4

# uncomment for composite PAL
#sdtv_mode=2

#uncomment to overclock the arm. 700 MHz is the default.
#arm_freq=800

# Uncomment some or all of these to enable the optional hardware interfaces
#dtparam=i2c_arm=on
#dtparam=i2s=on
#dtparam=spi=on

# Uncomment this to enable the lirc-rpi module
#dtoverlay=lirc-rpi

# Additional overlays and parameters are documented /boot/overlays/README

# Enable audio (loads snd_bcm2835)
#dtparam=audio=on
program_usb_boot_mode=1
start_x=0
gpu_mem=32
```

On master it shows 957MB

popcornmix, 2016-08-19:
Ping @ghollingworth

ghollingworth, 2016-08-21:
I've been pinged, will have a look next week when I'm back from holiday...

leidentech, 2016-08-24:
I'm seeing the same on a pi3 running off of USB:
root@kipi:/var/www# free -m
             total       used       free     shared    buffers     cached
Mem:           117        102         14          4          4         34
-/+ buffers/cache:         63         53
Swap:         2046          0       2046

Marsdo, 2016-08-26:
Also same issue when running my rpi3 off USB:
pi@marspi:~ $ free -m
             total       used       free     shared    buffers     cached
Mem:           117         88         28          4          7         49
-/+ buffers/cache:         31         85
Swap:           99          0         99

popcornmix, 2016-08-27:
Closing in favour of https://github.com/raspberrypi/firmware/issues/649
(which is reported in correct repo)

closed 2016-08-27 by popcornmix