Hexxeh/rpi-update #218: Raspbian version stays at 4.1.19+ after rpi-update

devinfluencer, 2016-08-25:
Hi,
I rpi-updated my Raspi several times, getting 4.4.19+ the last time. I did NOT forget to reboot.
My Linux version however shows 4.1.19+ with the consequence that my /lib/modules folder doesn't contain the right version files and so my CIFS subsystem doesn't work anymore.
Do I understand something wrong?
-Thomas

popcornmix, 2016-08-26:
Pi1, Pi2 or Pi3?
Report output of:
```
vcgencmd version
uname -a
ls -l /lib/modules
ls -l /boot/kernel*
cat /boot/.firmware_revision
sudo rpi-update
```

devinfluencer, 2016-08-26:
That's easy: It's a Pi1B

_pi@raspberrypi ~ $ vcgencmd version
Mar  9 2016 18:11:14
Copyright (c) 2012 Broadcom
version 3a754304b032a5298ee7889b179c667bbc75dec5 (clean) (release)
pi@raspberrypi ~ $ uname -a
Linux raspberrypi 4.1.19+ #853 Wed Mar 9 18:04:48 GMT 2016 armv6l GNU/Linux
pi@raspberrypi ~ $ ls -l /lib/modules
total 32
drwxr-xr-x 3 root root 4096 May  7  2015 3.18.7+
drwxr-xr-x 3 root root 4096 May  7  2015 3.18.7-v7+
drwxr-xr-x 3 root root 4096 Aug  8 23:29 4.4.13+
drwxr-xr-x 3 root root 4096 Aug  8 23:29 4.4.13-v7+
drwxr-xr-x 3 root root 4096 Aug 21 11:19 4.4.18+
drwxr-xr-x 3 root root 4096 Aug 21 11:19 4.4.18-v7+
drwxr-xr-x 3 root root 4096 Aug 25 18:37 4.4.19+
drwxr-xr-x 3 root root 4096 Aug 25 18:37 4.4.19-v7+
pi@raspberrypi ~ $ ls -l /boot/kernel*
-rw-r--r-- 1 root root 4226832 Aug 25 18:34 /boot/kernel7.img
-rw-r--r-- 1 root root 4126024 Aug 25 18:34 /boot/kernel.img
pi@raspberrypi ~ $ cat /boot/.firmware_revision
818a860cf690d64c62d3227ad9c36d5867a671c2
pi@raspberrypi ~ $ sudo rpi-update
 *** Raspberry Pi firmware updater by Hexxeh, enhanced by AndrewS and Dom
 *** Performing self-update
 *** Relaunching after update
 *** Raspberry Pi firmware updater by Hexxeh, enhanced by AndrewS and Dom
 *** Your firmware is already up to date
_

popcornmix, 2016-08-26:
Can you report output of:
```
vcgencmd get_config str
vcgencmd get_config int
mount
```

devinfluencer, 2016-08-26:
Thee you are:

pi@raspberrypi ~ $ vcgencmd get_config str
device_tree=-
pi@raspberrypi ~ $ vcgencmd get_config int
audio_pwm_mode=1
audio_sdm_mod_order=2
config_hdmi_boost=2
disable_auto_turbo=1
disable_commandline_tags=2
emmc_pll_core=1
force_pwm_open=1
framebuffer_ignore_alpha=1
framebuffer_swap=1
hdmi_force_cec_address=65535
ignore_lcd=1
init_uart_clock=0x2dc6c0
overscan_bottom=32
overscan_left=32
overscan_right=32
overscan_top=32
pause_burst_frames=1
program_serial_random=1
temp_limit=85
pi@raspberrypi ~ $ mount
/dev/mmcblk0p2 on / type ext4 (rw,noatime,data=ordered)
devtmpfs on /dev type devtmpfs (rw,relatime,size=218416k,nr_inodes=54604,mode=755)
sysfs on /sys type sysfs (rw,nosuid,nodev,noexec,relatime)
proc on /proc type proc (rw,relatime)
tmpfs on /dev/shm type tmpfs (rw,nosuid,nodev)
devpts on /dev/pts type devpts (rw,nosuid,noexec,relatime,gid=5,mode=620,ptmxmode=000)
tmpfs on /run type tmpfs (rw,nosuid,nodev,mode=755)
tmpfs on /run/lock type tmpfs (rw,nosuid,nodev,noexec,relatime,size=5120k)
tmpfs on /sys/fs/cgroup type tmpfs (ro,nosuid,nodev,noexec,mode=755)
cgroup on /sys/fs/cgroup/systemd type cgroup (rw,nosuid,nodev,noexec,relatime,xattr,release_agent=/lib/systemd/systemd-cgroups-agent,name=systemd)
cgroup on /sys/fs/cgroup/cpuset type cgroup (rw,nosuid,nodev,noexec,relatime,cpuset)
cgroup on /sys/fs/cgroup/cpu,cpuacct type cgroup (rw,nosuid,nodev,noexec,relatime,cpu,cpuacct)
cgroup on /sys/fs/cgroup/blkio type cgroup (rw,nosuid,nodev,noexec,relatime,blkio)
cgroup on /sys/fs/cgroup/devices type cgroup (rw,nosuid,nodev,noexec,relatime,devices)
cgroup on /sys/fs/cgroup/freezer type cgroup (rw,nosuid,nodev,noexec,relatime,freezer)
cgroup on /sys/fs/cgroup/net_cls type cgroup (rw,nosuid,nodev,noexec,relatime,net_cls)
systemd-1 on /proc/sys/fs/binfmt_misc type autofs (rw,relatime,fd=22,pgrp=1,timeout=300,minproto=5,maxproto=5,direct)
mqueue on /dev/mqueue type mqueue (rw,relatime)
debugfs on /sys/kernel/debug type debugfs (rw,relatime)
configfs on /sys/kernel/config type configfs (rw,relatime)
tmpfs on /run/user/1000 type tmpfs (rw,nosuid,nodev,relatime,size=44540k,mode=700,uid=1000,gid=1000)

popcornmix, 2016-08-26:
That is your problem. ``/boot`` should be mounted to FAT partition of sdcard. E.g.
```
pi@domnfs:~ $ mount |grep boot
/dev/mmcblk0p1 on /boot type vfat (rw,relatime,fmask=0022,dmask=0022,codepage=437,iocharset=ascii,shortname=mixed,errors=remount-ro)
```
From:
```
pi@domnfs:~ $ cat /etc/fstab 
proc            /proc           proc    defaults          0       0
/dev/mmcblk0p1  /boot           vfat    defaults          0       2
/dev/mmcblk0p2  /               ext4    defaults,noatime  0       1
# a swapfile is not a swap partition, no line here
#   use  dphys-swapfile swap[on|off]  for that
```
Are you using raspbian? Have you changed this?

devinfluencer, 2016-08-26:
Yes, raspbian.
I don*t think I changed anything. I*m a Windows man, so I not completely at home with Linux.
But I think it stopped to work one day.

popcornmix, 2016-08-26:
Post what is in your ``/etc/fstab``

devinfluencer, 2016-08-26:
pi@raspberrypi ~ $ cat /etc/fstab
proc /proc proc defaults 0 0 /dev/mmcblk0p1 /boot vfat defaults 0 2
/dev/mmcblk0p2 / ext4 defaults,noatime 0 1
# a swapfile is not a swap partition, so no using swapon|off from here on, use  dphys-swapfile swap[on|off]  for that
//linkstation/sources /sources cifs user=admin,password=repeater,sec=ntlmv2,rw,file_mode=0777,dir_mode=0777 0 0
//linkstation/share /media cifs user=admin,password=repeater,sec=ntlmv2,rw,file_mode=0777,dir_mode=0777 0 0
p

popcornmix, 2016-08-26:
Well you've obviously edited it. Add this line back in:
``/dev/mmcblk0p1  /boot           vfat    defaults          0       2``
reboot, then
```
sudo rm /boot/.firmware_revision
sudo rpi-update
sudo reboot
```
Should hopefully complete the update.

popcornmix, 2016-08-26:
Note:
```
proc /proc proc defaults 0 0 /dev/mmcblk0p1 /boot vfat defaults 0 2
```
Should be:
```
proc /proc proc defaults 0 0
/dev/mmcblk0p1 /boot vfat defaults 0 2
```
looks like you've removed the newline.

devinfluencer, 2016-08-27:
Looks like some "proc" installer left the newlines out, possibly that were two "proc" lines. Gonna look that up.
CIFS doen't work yet.

devinfluencer, 2016-08-27:
Looks like the webcam driver can't be loaded for the same reason. Missing /dev/video0, too.

popcornmix, 2016-08-27:
After reboot what does ``uname -a`` report/

devinfluencer, 2016-08-27:
Still the same.

popcornmix, 2016-08-27:
Please report output of all the commands I've requested previously

devinfluencer, 2016-08-27:
pi@raspberrypi ~ $ uname -a
Linux raspberrypi 4.1.19+ #853 Wed Mar 9 18:04:48 GMT 2016 armv6l GNU/Linux
pi@raspberrypi ~ $ vcgencmd version
Mar  9 2016 18:11:14
Copyright (c) 2012 Broadcom
version 3a754304b032a5298ee7889b179c667bbc75dec5 (clean) (release)
pi@raspberrypi ~ $ ls -l /lib/modules
total 32
drwxr-xr-x 3 root root 4096 May  7  2015 3.18.7+
drwxr-xr-x 3 root root 4096 May  7  2015 3.18.7-v7+
drwxr-xr-x 3 root root 4096 Aug  8 23:29 4.4.13+
drwxr-xr-x 3 root root 4096 Aug  8 23:29 4.4.13-v7+
drwxr-xr-x 3 root root 4096 Aug 21 11:19 4.4.18+
drwxr-xr-x 3 root root 4096 Aug 21 11:19 4.4.18-v7+
drwxr-xr-x 3 root root 4096 Aug 27 08:58 4.4.19+
drwxr-xr-x 3 root root 4096 Aug 27 08:58 4.4.19-v7+
pi@raspberrypi ~ $ ls -l /boot/kernel*
-rwxr-xr-x 1 root root 4056708 Mar 16 08:56 /boot/kernel7.img
-rwxr-xr-x 1 root root 3964132 Mar 16 08:56 /boot/kernel.img
pi@raspberrypi ~ $ cat /boot/.firmware_revision
ff93337300068d3cddd7f823bc77c4eb954357c9
pi@raspberrypi ~ $ sudo rpi-update
 *** Raspberry Pi firmware updater by Hexxeh, enhanced by AndrewS and Dom
 *** Performing self-update
 *** Relaunching after update
 *** Raspberry Pi firmware updater by Hexxeh, enhanced by AndrewS and Dom
This update bumps to rpi-4.4.y linux tree
Be aware there could be compatibility issues with some drivers
Discussion here:
https://www.raspberrypi.org/forums/viewtopic.php?f=29&t=144087
##############################################################
 *** Downloading specific firmware revision (this will take a few minutes)
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100   168    0   168    0     0    224      0 --:--:-- --:--:-- --:--:--   227
100 51.2M  100 51.2M    0     0   117k      0  0:07:26  0:07:26 --:--:--  118k
 *** Updating firmware
 *** Updating kernel modules
 *** depmod 4.4.19-v7+
 *** depmod 4.4.19+
 *** Updating VideoCore libraries
 *** Using HardFP libraries
 *** Updating SDK
 *** Running ldconfig
 *** Storing current firmware revision
 *** Deleting downloaded files
 *** Syncing changes to disk
 *** If no errors appeared, your firmware was successfully updated to 818a860cf690d64c62d3227ad9c36d5867a671c2
 *** A reboot is needed to activate the new firmware
pi@raspberrypi ~ $ sudo shutdown -r now
login as: pi
pi@rpi's password:

The programs included with the Debian GNU/Linux system are free software;
the exact distribution terms for each program are described in the
individual files in /usr/share/doc/*/copyright.

Debian GNU/Linux comes with ABSOLUTELY NO WARRANTY, to the extent
permitted by applicable law.
Last login: Sat Aug 27 15:35:39 2016 from thomas.fritz.box
pi@raspberrypi ~ $ vcgencmd get_config str
device_tree=-
pi@raspberrypi ~ $ vcgencmd get_config int
audio_pwm_mode=1
config_hdmi_boost=2
disable_auto_turbo=1
disable_commandline_tags=2
enable_uart=1
force_pwm_open=1
framebuffer_ignore_alpha=1
framebuffer_swap=1
hdmi_force_cec_address=65535
ignore_lcd=1
init_uart_clock=0x2dc6c00
overscan_bottom=32
overscan_left=32
overscan_right=32
overscan_top=32
pause_burst_frames=1
program_serial_random=1
temp_limit=85
pi@raspberrypi ~ $ mount
/dev/mmcblk0p2 on / type ext4 (rw,noatime,data=ordered)
devtmpfs on /dev type devtmpfs (rw,relatime,size=218260k,nr_inodes=54565,mode=755)
sysfs on /sys type sysfs (rw,nosuid,nodev,noexec,relatime)
proc on /proc type proc (rw,relatime)
tmpfs on /dev/shm type tmpfs (rw,nosuid,nodev)
devpts on /dev/pts type devpts (rw,nosuid,noexec,relatime,gid=5,mode=620,ptmxmode=000)
tmpfs on /run type tmpfs (rw,nosuid,nodev,mode=755)
tmpfs on /run/lock type tmpfs (rw,nosuid,nodev,noexec,relatime,size=5120k)
tmpfs on /sys/fs/cgroup type tmpfs (ro,nosuid,nodev,noexec,mode=755)
cgroup on /sys/fs/cgroup/systemd type cgroup (rw,nosuid,nodev,noexec,relatime,xattr,release_agent=/lib/systemd/systemd-cgroups-agent,name=systemd)
cgroup on /sys/fs/cgroup/cpu,cpuacct type cgroup (rw,nosuid,nodev,noexec,relatime,cpu,cpuacct)
cgroup on /sys/fs/cgroup/blkio type cgroup (rw,nosuid,nodev,noexec,relatime,blkio)
cgroup on /sys/fs/cgroup/memory type cgroup (rw,nosuid,nodev,noexec,relatime,memory)
cgroup on /sys/fs/cgroup/devices type cgroup (rw,nosuid,nodev,noexec,relatime,devices)
cgroup on /sys/fs/cgroup/freezer type cgroup (rw,nosuid,nodev,noexec,relatime,freezer)
cgroup on /sys/fs/cgroup/net_cls type cgroup (rw,nosuid,nodev,noexec,relatime,net_cls)
systemd-1 on /proc/sys/fs/binfmt_misc type autofs (rw,relatime,fd=22,pgrp=1,timeout=300,minproto=5,maxproto=5,direct)
mqueue on /dev/mqueue type mqueue (rw,relatime)
debugfs on /sys/kernel/debug type debugfs (rw,relatime)
fusectl on /sys/fs/fuse/connections type fusectl (rw,relatime)
configfs on /sys/kernel/config type configfs (rw,relatime)
/dev/mmcblk0p1 on /boot type vfat (rw,relatime,fmask=0022,dmask=0022,codepage=437,iocharset=ascii,shortname=mixed,errors=remount-ro)
binfmt_misc on /proc/sys/fs/binfmt_misc type binfmt_misc (rw,relatime)
tmpfs on /run/user/1000 type tmpfs (rw,nosuid,nodev,relatime,size=44512k,mode=700,uid=1000,gid=1000)
pi@raspberrypi ~ $ cat /etc/fstab
proc /proc proc defaults 0 0
/dev/mmcblk0p1 /boot vfat defaults 0 2
/dev/mmcblk0p2 / ext4 defaults,noatime 0 1
//linkstation/sources /sources cifs user=XXX,password=XXX,sec=ntlmv2,rw,file_mode=0777,dir_mode=0777 0 0
//linkstation/share /media cifs user=XXX,password=XXX,sec=ntlmv2,rw,file_mode=0777,dir_mode=0777 0 0
pi@raspberrypi ~ $ uname -a
Linux raspberrypi 4.4.19+ #906 Tue Aug 23 15:40:04 BST 2016 armv6l GNU/Linux

popcornmix, 2016-08-27:
Okay, so /boot is mounted and kernel is up to date. That is good. What does:
``sudo modprobe cifs``
report?
If that doesn't work does ``sudo depmod`` help?
What about 
``ls -l /lib/modules/4.4.19+/kernel/fs/cifs/cifs.ko``

devinfluencer, 2016-08-27:
pi@raspberrypi ~ $ sudo modprobe cifs
pi@raspberrypi ~ $ sudo depmod
pi@raspberrypi ~ $ ls -l /lib/modules/4.4.19+/kernel/fs/cifs/cifs.ko
-rw-r--r-- 1 root root 524256 Aug 27 16:38 /lib/modules/4.4.19+/kernel/fs/cifs/cifs.ko

Looks good now, /dev/video0 is there again, w1-gpio too, NAS shares too.
Thank you!

Was there some problem with the kernel distribution?

popcornmix, 2016-08-28:
No, the only problem I can see was in your ``/etc/fstab`` file which meant rpi-update wasn't able to update kernel (from /boot) but was able to update ``/lib/modules``.
I believe that problem was unique to you.
Feel free to close this issue if you are updated correctly.

devinfluencer, 2016-08-28:
Sounds plausible.
Thanks again.

closed 2016-08-28 by devinfluencer