Hexxeh/rpi-update #219: Issue on reboot rpi-update systemd #ArchLinux Distrib

Ben00it, 2016-08-27:
Hi, 

Many thanks for your script I used since years on all my rpi. 
But recently on all my rpi 2 and 3 (under arch linux) I encountered the same issue. 

When I reboot it after made a full system upgrade (sudo pacman -Syyu) + rpi-update, I got the service systemd load failed. 
Failed at step SECCOMP 228 spawning

```
[Sun Jul 31 05:34:35 2016] systemd[1]: systemd-logind.service: Main process exited, code=exited, status=228/SECCOMP
[Sun Jul 31 05:34:35 2016] systemd[1]: Failed to start Login Service.
[Sun Jul 31 05:34:35 2016] systemd[1]: systemd-logind.service: Unit entered failed state.
[Sun Jul 31 05:34:35 2016] systemd[1]: systemd-logind.service: Failed with result 'exit-code'.
[Sun Jul 31 05:34:35 2016] systemd[1]: systemd-logind.service: Service has no hold-off time, scheduling restart.
[Sun Jul 31 05:34:35 2016] systemd[1]: Stopped Login Service.
[Sun Jul 31 05:34:35 2016] systemd[1]: Starting Login Service...
[Sun Jul 31 05:34:35 2016] systemd[935]: systemd-logind.service: Failed at step SECCOMP spawning /usr/lib/systemd/systemd-logind: Invalid argument
```

On one of them I could restore it by deleting those files, then run again sudo pacman -Syyu: 

```
sudo rm  -rf /boot/overlays/audremap.dtbo
sudo rm -rf /boot/overlays/mcp23s17.dtbo
```
I said one of them (rpi2) because I could restore the network with dhclient that I installed separately previously. 
On other one (rpi3) I didn't install dhclient, only native dhcpd arch based on systemd, but it broke, so impossible to edit symlink /etc/resolv.conf... no network to run a new pacman upgrade... 

NB : I ran a pacman -Syyu just before rpi-update.

It's a bit embarrassing, any way to fix it in the future ? 

Many thanks for your feedback, and keep going on.

Ben00it, 2016-08-29:
Hi, 
Problem solved. 

I share the fix : 

From rpi2 (working): 

```
sudo pacman -Sw dhclient-4.3.4-2-armv7h.pkg.tar.xz
scp /var/cache/pacman/pkg/dhclient-4.3.4-2-armv7h.pkg.tar.xz user@rpi3:/home/user/
```

On fault rpi3 (not working)
```
sudo pacman -U dhclient-4.3.4-2-armv7h.pkg.tar.xz
sudo chmod 744 /etc/resolv.conf 
sudo dhclient eth0
sudo chmod 644 /etc/resolv.conf
sudo pacman -Syyu
sudo pacman -S linux-raspberrypi
sudo pacman -S raspberrypi-firmware
sudo pacman -S raspberrypi-firmware-tools
sudo pacman -S linux-raspberrypi-headers
sudo reboot
```
And then all good.

popcornmix, 2016-08-29:
Yes, rpi-update is designed for updating raspbian.
You should check with maintainers of other distributions if using rpi-update is safe.
In general if they use a custom kernel (which I believe Arch does) then rpi-update should not be used.
Yes, I believe Arch provides its own firmware package that is updated quite regularly and that should be used.

Ben00it, 2016-08-29:
Hi Popcornmix. 
Many thanks for your update. 
That's right, I didn't notice those packages on arch. But you're right, here some extract from: http://raspberrypi.stackexchange.com/questions/164/how-do-i-update-software-and-firmware

Updating software - Arch Linux
_The software must be maintained. The advantage of Arch Linux over Debian here is that Arch Linux manages the Raspberry Pi's firmware within the package management system. To update, just run sudo `pacman -Syu`
If you're running Arch Linux then the updates will be a part of your regular pacman -Syu as they exist in the official RPi repositories._

Anyway those who've made the same mistake as me will found out how to fix it on a heavy rpi in PRODUCTION :)

Many thanks for your feedback popcormix. 
Keep it up !

pbatard, 2016-08-30:
> Failed at step SECCOMP 228 spawning

For the record, this looks like a SECCOMP detection bug with Systemd, which will bring down `jounald`, `networkd`, `logind` and a bucnh of other services, and which also got picked by Debian. See https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=832713

Ben00it, 2016-08-30:
Yep, and believe me I googled a lot. Since the systemd 231 it's appear SECCOMP is require in the kernel. 
There are a lot of similar issue : It's because the kernel is not built with SECCOMP support, but libseccomp does not expose a check to systemd 231 for determining if it's actually enabled at runtime.
https://archlinuxarm.org/forum/viewtopic.php?f=9&t=10677
https://github.com/systemd/systemd/issues/3882

Ruffio, 2017-04-05:
@Ben00it can this issue be closed?

Ben00it, 2017-04-06:
Hi, of course. And many thanks for the help. 
I confirm on Archlinux, just run pacman -Syu. 
Work perfectly since.

closed 2017-04-06 by Ben00it