Hexxeh/rpi-update #220: opensuse Tumbleweed on Raspberry 2: getting `/opt/vc/bin/vcgencmd: error while loading shared libraries: libvchiq_arm.so: cannot open shared object file: No such file or directory`

jpluimers, 2016-08-30:
I'm running an almost pristine [openSUSE-Tumbleweed-ARM-JeOS-raspberrypi2.armv7l.raw.xz](http://download.opensuse.org/repositories/devel:/ARM:/Factory:/Contrib:/RaspberryPi2/images/openSUSE-Tumbleweed-ARM-JeOS-raspberrypi2.armv7l.raw.xz) on a [Raspberry Pi 2 Model B v1.1](https://www.raspberrypi.org/products/raspberry-pi-2-model-b/)

```
cd /root/bin
wget https://raw.githubusercontent.com/Hexxeh/rpi-update/master/rpi-update 
chmod 755 rpi-update
./rpi-update
reboot
```
(All above steps successful)

I get this:

```
statler:/opt/vc # /opt/vc/bin/vcgencmd version
/opt/vc/bin/vcgencmd: error while loading shared libraries: libvchiq_arm.so: cannot open shared object file: No such file or directory
statler:/opt/vc # locate libvchiq_arm.so
/opt/vc/lib/libvchiq_arm.so
statler:/opt/vc # ldd /opt/vc/bin/vcgencmd
        linux-vdso.so.1 (0x7ecf1000)
        libvchiq_arm.so => not found
        libvcos.so => not found
        libpthread.so.0 => /lib/libpthread.so.0 (0x76f5e000)
        libdl.so.2 => /lib/libdl.so.2 (0x76f48000)
        librt.so.1 => /lib/librt.so.1 (0x76f31000)
        libc.so.6 => /lib/libc.so.6 (0x76de8000)
        /lib/ld-linux-armhf.so.3 (0x54ae9000)
statler:/opt/vc # locate libvcos.so
/opt/vc/lib/libvcos.so
statler:/opt/vc # ldd /opt/vc/lib/libvchiq_arm.so
ldd: warning: you do not have execution permission for `/opt/vc/lib/libvchiq_arm.so'
        linux-vdso.so.1 (0x7eee3000)
        libvcos.so => not found
        libpthread.so.0 => /lib/libpthread.so.0 (0x76ec8000)
        libdl.so.2 => /lib/libdl.so.2 (0x76eb3000)
        librt.so.1 => /lib/librt.so.1 (0x76e9b000)
        libc.so.6 => /lib/libc.so.6 (0x76d52000)
        /lib/ld-linux-armhf.so.3 (0x54b46000)
statler:/opt/vc # ldd /opt/vc/lib/libvcos.so
ldd: warning: you do not have execution permission for `/opt/vc/lib/libvcos.so'
        linux-vdso.so.1 (0x7ee3d000)
        libpthread.so.0 => /lib/libpthread.so.0 (0x76f28000)
        libdl.so.2 => /lib/libdl.so.2 (0x76f13000)
        librt.so.1 => /lib/librt.so.1 (0x76efc000)
        libc.so.6 => /lib/libc.so.6 (0x76db2000)
        /lib/ld-linux-armhf.so.3 (0x54ba6000)
statler:/opt/vc # echo $PATH
/sbin:/usr/sbin:/usr/local/sbin:/root/bin:/usr/local/bin:/usr/bin:/bin:/usr/games:/opt/vc/lib:/opt/vc/bin
```

Any idea what I'm missing?

popcornmix, 2016-08-30:
rpi-update is designed for raspbian. You should really ask opensuse maintainers if it is suitable for their distribution.
Did ``/opt/vc/lib`` exist before you ran rpi-update? 
``opt/vc/lib`` need to be added to library path somehow. raspbian does it with:
```
cat /etc/ld.so.conf.d/00-vmcs.conf 
/opt/vc/lib
```
but use of ``LD_LIBRARY_PATH`` and no doubt other ways are possible. Does:
```LD_LIBRARY_PATH=/opt/vc/lib /opt/vc/bin/vcgencmd version```
work?

jpluimers, 2016-08-30:
Found that out myself as well that this works: 

```
LD_LIBRARY_PATH=$LD_LIBRARY_PATH:/opt/vc/lib/
export LD_LIBRARY_PATH
/opt/vc/bin/vcgencmd version
```

There was no `/opt/vc/lib` before installing `rpi-update` and I totally forgot that in the Linux world `PATH` and `LD_LIBRARY_PATH` are two different things. The #opensuse-arm IRC channel reminded me about that though (:

popcornmix, 2016-08-30:
I've made the README text for rpi-update more explicit that it is designed for raspbian and should only be used on other distributions if the maintainers report it is safe.

I suspect you shouldn't be using it with opensuse.

jpluimers, 2016-08-30:
I was assuming `rpi-update` would update the embedded firmware (GPU, etc) only, not mess with kernels and such.

popcornmix, 2016-08-30:
It updates the gpu firmware, the kernel and the userland (user libs for communicating with the firmware).
Often changes are required in all three to support a new gpu feature.
They should ideally be updated together.

jpluimers, 2016-08-30:
This opens even more questions for me:

- What if for some reason I'd only want the userland binaries or the firmware?
- How would one know which kernel is compatible with what firmware / userland binaries?

Thanks for all the help so far.

popcornmix, 2016-08-30:
Obviously if a new feature required a change to say, firmware and kernel or to firmware and userland then you won't benefit from that feature if you only update one of them.

In general we try to handle mismatched versions as cleanly as possible - we think about whether the behaviour is sensible if one component is updated without another, and in general if you don't use the new feature things will continue to work and if you do try then errors are reported rather than a fatal condition.

But some bigger changes do require some cooperation. The 4.4 kernel tree only works with device tree which is injected from firmware. A 4.4 kernel won't boot without a firmware that is new enough to create the device tree. That sort of change is pretty rare.

In general we aim that you can always update gpu firmware (bootcode.bin/start.elf/fixup.dat) safely. Updating kernel or userland will normally work, but occasionally will require the firmware to be at least as new.

jpluimers, 2016-08-30:
I verified the files that go with `opensuse Tumbleweed` match with what `rpi-update` installed:

```
5bbb5f25dfe16f2871d5defcc48ac9ff  /boot/efi/bootcode.bin
5bbb5f25dfe16f2871d5defcc48ac9ff  /boot/vc/bootcode.bin
03f6b11cdc10260ebd1e8edfd4897356  /boot/efi/start.elf
e99d0923fea1914db8d5a87a2bb1cbf0  /boot/vc/start.elf
562dc460ce68eca4e017c6666ee2dbb7  /boot/efi/fixup.dat
363b45a30518c3b8cada40ef7953b5d4  /boot/vc/fixup.dat
03f6b11cdc10260ebd1e8edfd4897356  /boot/efi/start.elf
e99d0923fea1914db8d5a87a2bb1cbf0  /boot/vc/start.elf
```

These are missing in opensuse:

```
md5sum: /boot/start_cd.elf: No such file or directory
md5sum: /boot/start_db.elf: No such file or directory
md5sum: /boot/start.elf: No such file or directory
md5sum: /boot/start_x.elf: No such file or directory
```

On opensuse, `/opt` is empty, that's why I think a "userland-only" or similar feature would be nice. 

I've made a proposal called `SKIP_FIRMWARE` that I think will do and post that as a pull-request for this issue # shortly.

Ben00it, 2016-08-30:
Hi  jpluimers, 
Have you check there : http://software.opensuse.org/download.html?project=hardware&package=raspberrypi-firmware

jpluimers, 2016-08-30:
I just checked and both these rpms are installed by default on the [openSUSE-Tumbleweed-ARM-JeOS-raspberrypi2.armv7l.raw.xz](http://download.opensuse.org/repositories/devel:/ARM:/Factory:/Contrib:/RaspberryPi2/images/openSUSE-Tumbleweed-ARM-JeOS-raspberrypi2.armv7l.raw.xz) image:

```
# rpm -qa | grep raspberrypi-firmware
raspberrypi-firmware-2016.08.23-1.1.noarch
raspberrypi-firmware-config-rpi2-2016.08.23-1.1.noarch
```

But these do not include the userland binaries.

jpluimers, 2016-08-30:
If you want, I could post a small update to this pull request that also gets rid of this message in the `SKIP_FIRMWARE=1 UPDATE_SELF=0 ./rpi-update` case:

```
 *** A reboot is needed to activate the new firmware
```

Ruffio, 2020-12-25:
@jpluimers can this issue be closed?