Hexxeh/rpi-update #222: Add documentation for downgrading firmware / "un-install" rpi-update

crw, 2016-09-19:
Howdy! I kind of blindly ran rpi-update without *ahem* closely reading the docs. I did not realize at the time that the apt-get updates would update rpi firmware to the latest stable before I did this. My request is, could you please add a section to the README.md covering how to uninstall the changes made by rpi-update and get back on the latest stable firmware? Thanks!

Ruffio, 2017-01-22:
@crw You can't 'uninstall' but you can install ealier firmware by using Option: https://github.com/Hexxeh/rpi-update#options

popcornmix, 2017-01-22:
```sudo apt-get install --reinstall raspberrypi-bootloader raspberrypi-kernel```
will get you back to latest stable (apt-get) firmware/kernel.

lemassykoi, 2017-03-29:
I used rpi-update to update to 4.9, unfortunately, my DomoticZ install hangs up because of issue with GPIO.
To revert, I used `sudo BRANCH=stable rpi-update` and it's all revert back nicely.

Ruffio, 2017-04-05:
@crw can this be closed?

crw, 2017-04-05:
Yep.

closed 2017-04-05 by crw