Hexxeh/rpi-update #224: rpi-update breaks latest Raspbian PIXEL login after update

mitchhh22, 2016-09-29:
I installed a fresh copy of the new Raspbian released a few days ago. Everything works fine until I update the firmware with rpi-update. 
Now the Desktop login is broken. It tries to login but just flashes the screen and returns back to the login prompt.

jgwatkins, 2016-10-10:
Same here.  Any luck?

popcornmix, 2016-10-10:
I've tried rpi-update and it didn't cause any issue for me. @spl237 also doesn't see the issue.
First does reverting firmware back to version from last raspbian jessie release restore behaviour?
```sudo rpi-update 21505f3fb5d8e2a5abbf95baf441a7a6415d4a82```

jgwatkins, 2016-10-12:
Thx Popcornmix!

Turns out i had a corrupt .Xauthority file.  I just deleted it and was able to login.  Appreciate your help! :)

jgwatkins, 2016-10-12:
to mitchhh22:

I was able to fix this problem doing to following:

When the "greeter" screen is up and doing its endless login loop as you described:

SSH into the RPI with another local machine
type:
`ls -a`
look for a file named .Xauthority
delete .Xauthority `rm .Xauthority`

Hope this works

popcornmix, 2016-10-12:
Pretty sure this issue is unrelated to rpi-update so closing.
If you have evidence for .Xauthority being corrupted (that isn't the fault of your own scripts) then please report here: https://github.com/RPi-Distro/raspberrypi-ui-mods

closed 2016-10-12 by popcornmix