Hexxeh/rpi-update #225: [feature request] Ask before updating

Firesphere, 2016-10-06:
When updating, it would be nice if the updater said something like
```
Updating from version 4.4.20+ to 4.4.23+; continue? [Y/n]
```
So as a user, I know what is going to happen if I say yes.

Ruffio, 2016-10-06:
@Firesphere I'm not the developer of this, but I don't see the usage of this. Why would you run the rpi-update if you are not interested in doing an update?

Firesphere, 2016-10-06:
It doesn't need to be the question. But at least inform the user before going forward. Same as `apt-get upgrade` does. It tells you the changes and asks if you want to do it.

I would be interested in it, as, for example, the latest update I ran, I noticed a notice saying "this will bump the kernel to 4.4.x, which might cause problems"
It would be nice to give the option to cancel to prevent those problems.

Ruffio, 2016-10-06:
Hmmm, I don't quite agree with your comparison. apt-get upgrade goes for all packages. You should only execute rpi-update if you really want to update the firmware and you know what you are doing. That is the same if you really want to update the firmware on your PC.

The notice about bumping to kernel 4.4.x has been there a really long time, I think it is a year or something like that.

IMHO no ordinary user should use the rpi-update, just like no ordinary user should upgrade firmware on PC.

Ruffio, 2016-10-06:
Xref: https://github.com/Hexxeh/rpi-update/pull/201

Firesphere, 2016-10-06:
I disagree with that, as rpi-update is even advertised on multiple websites, as the go-to way to upgrade your kernel.
And while apt will run all available updates, the kernel is still the core. At least informing the person running the command is at the very worst, common courtesy.

There are a lot of people, who blindly follow a website like Lifehacker or whoever is an authority and run rpi-upgrade without knowing the implications.

Ruffio, 2016-10-06:
As I said I'm not the developer, so it is not my decisions. The Xref is to a PR similar to this and that hasn't been merged...

popcornmix, 2016-10-06:
See the README on:
https://github.com/Hexxeh/rpi-update

> Even on Raspbian you should only use this with a good reason.

Many updates here don't bump the kernel version, so a message like:
> Updating from version 4.4.20+ to 4.4.23+; continue? [Y/n]

doesn't make sense.

Gadgetoid, 2017-02-21:
The first line of the README is rather misleading, and even to this day I see people who don't understand what rpi-update is, running it in order to get updates for their Pi.

If this statement was true in the past, it certainly isn't anymore.

```
An easier way to get the latest firmware for your Raspberry Pi
```

I spent a lot of effort actively warning people against rpi-update and encouraging websites to remove references to it as the canonical source of updates. This was to stem a tide of support requests from people who had basically broken their Raspbian install with pre-release firmware/kernel releases and couldn't get anything to work.

I think there's a rift between the people actually experienced enough to use this utility, and the people who have to deal with the fallout from it being used unwittingly and adding a warning would certainly help to close that gap.

That said, the problem is pretty minimal these days, but I'd wholly encourage attention to this Issue and the PR linked because it could save someone a real headache in the future.

RogerWeihrauch, 2017-07-24:
@Ruffio (since I know you're not the developer; but to whom to report to?)
Me for myself, I would also like to get a hint on what 'rpi-update' may be going to do after having entered: 'pi@raspberrypi:~ $ 'sudo JUST_CHECK=1 rpi-update''; and not only get the message that i 1st have to execute it before getting information on what will be upgraded/which version will be installed (see: We're running for the first time)
 *** Raspberry Pi firmware updater by Hexxeh, enhanced by AndrewS and Dom
 *** Performing self-update
 *** Relaunching after update
 *** Raspberry Pi firmware updater by Hexxeh, enhanced by AndrewS and Dom
 *** We're running for the first time
pi@raspberrypi:~ $

Ruffio, 2017-07-24:
@popcornmix is the developer. Im just a uset like you...

popcornmix, 2017-07-25:
``sudo JUST_CHECK=1 rpi-update`` will show you the list of commits between the previous rpi-update and the new rpi-update firmware/kernel.
If you are running rpi-update for the first time it won't have anything to print.

moodeaudio, 2018-02-24:
I noticed that the bump to 4.14 branch requires YN confirmation to proceed which is fine but I'd propose either an env var or  a -y param (similar to apt-get) to support expert, non-interactive installs.

lurch, 2019-06-08:
@moodeaudio see #275

Ruffio, 2020-12-25:
@Firesphere can this be closed?

Firesphere, 2020-12-25:
Yes, as I think it's actually been implemented exactly the way I intended, a small confirmation system, that tells you what's gonna happen.

I agree with @moodeaudio , that a `-y` would be a next nice-to-have, but to me personally, I don't care that much about that feature. I'm absolutely happy with the question as it is right now. And also, my bad, I somewhat forgot I made this PR until you commented @Ruffio !

Very much cheers and have a great holiday/christmas/new year/etc. :)

closed 2020-12-25 by Firesphere