Hexxeh/rpi-update #228: rpi-update is currently broken

giminni, 2017-01-13:
i cannot use "sudo rpi-update". rpi-udate file will be substituted by a varnish cache server html error page

Error 503 Backend.max_conn reached!

This is an infrastructure error,
Please check for this error in rpi-update code notify maintainer and prevent rewrite of file.

Thx

popcornmix, 2017-01-13:
github (which rpi-update uses) was down a little while ago - seems to be working for me now.
Try again.

giminni, 2017-01-13:
@popcornmix 
Yes indeed it works now, 
but try to catch this error otherwise a user get an html error page that rewrites rpi-update :-(

popcornmix, 2017-01-13:
We do check for an error returned from github:
https://github.com/Hexxeh/rpi-update/blob/master/rpi-update#L50

But it appears they didn't return an error here. Do you still have the bad ``rpi-update`` file?

giminni, 2017-01-13:
no I do not have the file

It was a proper formatted HTML error page from varnish cache server stating
Error 503 Backend.max_conn reached!

popcornmix, 2017-01-13:
It looks like ``-f`` will cause a fail return code for HTTP errors >= 400.
I've added this option which hopefully will avoid this issue in future.
Need to wait for a github failure again to be sure...

giminni, 2017-01-13:
all rpi-update users will love it .... finally 👍

kmscode, 2017-01-13:
I had problems as well... I don't have the the file either, but this is the output that I received:
```
sudo rpi-update
 *** Raspberry Pi firmware updater by Hexxeh, enhanced by AndrewS and Dom
 *** Performing self-update
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100   448  100   448    0     0     29      0  0:00:15  0:00:15 --:--:--   115
 *** Relaunching after update
/usr/bin/rpi-update: line 2: syntax error near unexpected token `newline'
/usr/bin/rpi-update: line 2: `<?xml version="1.0" encoding="utf-8"?>'
```

Seems to be working now. 

Since it seems that a few users were getting html files for whatever reason, maybe an additional check after a "successful" download - something like this:
```
file ${_tempFileName} | grep "shell script"
if [ $? -ne 0 ]; then
   echo " !!! Failed to download update for rpi-update!"
   echo " !!! Check something..."
   exit 1
fi
```

kmscode, 2017-01-13:
Just saw the comment on "-f" - likely a cleaner solution, but won't fix other unknown scenarios where a file is "successfully" downloaded, but is not the proper file (or file type). However, I can't think of a fully plausable scenario with return codes less than 400... Unless a 302 some how mucks things up...

popcornmix, 2017-01-13:
Lets see if there any issues with the current fix. In theory it will solve the problem.

darkwinternight, 2017-03-15:
The "correct" fix would be to check for the file hash before actually replacing `rpi-update`. Or alternatively to use the GitHub URL with the `.patch` extension instead of the `raw.githubusercontent.com` URL and then to apply the patch using `patch`.

Ruffio, 2017-05-16:
@giminni can this be closed?

popcornmix, 2017-05-30:
Closing due to lack of activity.
Please re-open if there is still an issue.

closed 2017-05-30 by popcornmix