Hexxeh/rpi-update #229: [Feature Request]Possibility of installing kernel8.img(arm64) though rpi-update

Electron752, 2017-02-01:
This is way future thinking, but I was wondering about the possibility of having rpi-update distribute test versions of kernel8.img(arm64 version of kernel).

Of course it would need to display a gazillion warnings about it being a test version and that by trying it your totally at your own risk. etc...

I thought about making binary versions available in the past, but I wasn't sure about the legal implications of doing something like that.  And I'm not so sure I want to build even test versions on my personal computers although that's a possibility.

For local use, I have debian packages of the firmware and personal linux kernels.  The issue is that apt doesn't work so well since it doesn't like the FAT file system that the bootloader/firmware files need to be located on.

popcornmix, 2017-02-01:
I think at this point in time that publishing arm64 kernels will make them seem more officially supported than they really are. I think it is something we would be willing when the working features are broadly comparable with the arm32 kernel (e.g. having the hello_pi gpu demos working).

But if you want to encourage more testing of the arm64 kernel, then you can clone rpi-firmware and push builds to your clone. Then use:
```sudo REPO_URI=https://github.com/Electron752/rpi-firmware rpi-update```
to distribute them

Electron752, 2017-02-07:
Thanks,

At this point I think most of the changes to get a very functional arm64 kernel working are floating around.  And the know how is certainly available.

I'm considering setting up an alternative github source tree for the RPI 3 and reduce trying to contribute to the main upstream or downstream trees.   I really don't want to step on any toes by doing that.  Do you know of any problems doing that from a bootstrapping point of view?  I really wouldn't want a future official firmware update to suddenly stop supporting 64 bit boot.

I don't want to compete with the official trees, just augment them.   And unfortunately, it's a bit too difficult to make changes in the established trees.

popcornmix, 2017-02-07:
We have no issues with you (or others) providing custom kernels and using rpi-update to distribute them (that is why ``REPO_URI`` was added to rpi-update).

> I really wouldn't want a future official firmware update to suddenly stop supporting 64 bit boot.

Nor do we. If that ever happens let us know and we'll sort it out.
At the moment 64-bit kernels are very much for experimental users, but they should become more stable and feature in time and with increasing numbers of users.
Eventually I imagine there will be more official support, so encouraging use and testing is good.

closed 2017-02-08 by Electron752