Hexxeh/rpi-update #230: Linux 4.9.11 source download is broken

rascol, 2017-02-25:
Attempted to download Linux kernel v4.9.11 for Raspberry Pi using the rpi-source script. The archive is incomplete. This same result was obtained doing a zip download when v4.9.11 was current. This is the output from rpi-source running on the Raspberry Pi:

 *** gcc version check: OK

 *** rpi-update: https://github.com/Hexxeh/rpi-firmware

 *** Firmware revision: 3e1c3e9ba2b948e6023114b9633223363da78679

 *** Linux source commit: 204050d0eafb565b68abf512710036c10ef1bd23

 *** Download kernel source
--2017-02-25 13:31:18--  https://github.com/raspberrypi/linux/archive/204050d0eafb565b68abf512710036c10ef1bd23.tar.gz
Resolving github.com (github.com)... 192.30.253.112, 192.30.253.113
Connecting to github.com (github.com)|192.30.253.112|:443... connected.
HTTP request sent, awaiting response... 302 Found
Location: https://codeload.github.com/raspberrypi/linux/tar.gz/204050d0eafb565b68abf512710036c10ef1bd23 [following]
--2017-02-25 13:31:19--  https://codeload.github.com/raspberrypi/linux/tar.gz/204050d0eafb565b68abf512710036c10ef1bd23
Resolving codeload.github.com (codeload.github.com)... 192.30.253.120, 192.30.253.121
Connecting to codeload.github.com (codeload.github.com)|192.30.253.120|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: unspecified [application/x-gzip]
Saving to: ‘/home/pi/rpi/linux-204050d0eafb565b68abf512710036c10ef1bd23.tar.gz’

/home/pi/rpi/linux-204050d0eafb565b     [   <=>                                                              ] 138.16M   708KB/s   in 3m 20s 

2017-02-25 13:34:40 (707 KB/s) - ‘/home/pi/rpi/linux-204050d0eafb565b68abf512710036c10ef1bd23.tar.gz’ saved [144868917]

 *** Unpack kernel source
...............................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................
gzip: stdin: invalid compressed data--format violated
tar: Unexpected EOF in archive
tar: Unexpected EOF in archive
tar: Error is not recoverable: exiting now
Traceback (most recent call last):
  File "/usr/bin/rpi-source", line 345, in <module>
    sh("cd %s && tar --checkpoint=100 --checkpoint-action=dot -xzf %s" % (args.dest, linux_tar))
  File "/usr/bin/rpi-source", line 85, in sh
    subprocess.check_call(cmd, shell=True)
  File "/usr/lib/python2.7/subprocess.py", line 540, in check_call
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command 'cd /home/pi/rpi && tar --checkpoint=100 --checkpoint-action=dot -xzf /home/pi/rpi/linux-204050d0eafb565b68abf512710036c10ef1bd23.tar.gz' returned non-zero exit status 2

rascol, 2017-02-27:
The download is OK. Error was in the copy of Raspian being used. Sorry for the inconvenience.

hrj, 2017-03-23:
@rascol You can close the issue to reduce clutter.

closed 2017-03-23 by popcornmix