Hexxeh/rpi-update #231: Suspicious host needed for update

pdkovacs, 2017-02-26:
When trying to update my firmware with an Ubuntu Mate distribution, `rpi-update` is trying to connect to port `443` of `151.101.112.133 ` which resolves to `pri.authdns.ripe.net`. Is it safe to allow this connection? I am asking this in particular, because another of my hosts with the same Ubuntu distribution successfully completes the firmware update (through `rpi-update`) without wanting to connect to said host.

```
root@pici:~# uname -a
Linux pici 4.4.43-v7+ #948 SMP Sun Jan 15 22:20:07 GMT 2017 armv7l armv7l armv7l GNU/Linux
```

darkwinternight, 2017-03-15:
This should be the `raw.githubusercontent.com` host meaning it's safe.

popcornmix, 2017-03-15:
Looks like a github IP address (they have many), so no problem.

pdkovacs, 2017-03-15:
Why do you think it has anything to do with github:
```
[pkovacs@lapos webpack-boilerplate]$ dig -x 151.101.112.133

; <<>> DiG 9.8.3-P1 <<>> -x 151.101.112.133
;; global options: +cmd
;; Got answer:
;; ->>HEADER<<- opcode: QUERY, status: NXDOMAIN, id: 33527
;; flags: qr rd ra; QUERY: 1, ANSWER: 0, AUTHORITY: 1, ADDITIONAL: 0

;; QUESTION SECTION:
;133.112.101.151.in-addr.arpa.	IN	PTR

;; AUTHORITY SECTION:
151.in-addr.arpa.	83	IN	SOA	pri.authdns.ripe.net. dns.ripe.net. 1489583801 3600 600 864000 3600

;; Query time: 17 msec
;; SERVER: 192.168.1.3#53(192.168.1.3)
;; WHEN: Wed Mar 15 21:06:10 2017
;; MSG SIZE  rcvd: 106
```
?

popcornmix, 2017-03-15:
Where I am raw.github.com resolves to 151.101.16.133 which seems coincidentally similar.
Searching google for 151.101.16.133 and github shows a number of hits.

darkwinternight, 2017-03-15:
Sometimes I'm really surprised. You know how to use `dig` but don't try to go to http://151.101.112.133

popcornmix, 2017-03-15:
The IP address for raw.github.com I see is 151.101.16.133 which gives the same dig lookup as your suspicious IP address.

```
$ dig -x 151.101.16.133

; <<>> DiG 9.10.3-P4-Ubuntu <<>> -x 151.101.16.133
;; global options: +cmd
;; Got answer:
;; ->>HEADER<<- opcode: QUERY, status: NXDOMAIN, id: 8891
;; flags: qr rd ra ad; QUERY: 1, ANSWER: 0, AUTHORITY: 1, ADDITIONAL: 1

;; OPT PSEUDOSECTION:
; EDNS: version: 0, flags:; udp: 4096
;; QUESTION SECTION:
;133.16.101.151.in-addr.arpa.	IN	PTR

;; AUTHORITY SECTION:
151.in-addr.arpa.	3011	IN	SOA	pri.authdns.ripe.net. dns.ripe.net. 1489583801 3600 600 864000 3600

;; Query time: 0 msec
;; SERVER: 127.0.1.1#53(127.0.1.1)
;; WHEN: Wed Mar 15 20:18:33 GMT 2017
;; MSG SIZE  rcvd: 116
```