Hexxeh/rpi-update #236: no updating

monouser7dig, 2017-03-14:
thirst, thanks for the tool!

however, running it on my raspbian install does not upgrade the kernel
I'm currently on 4.4.50-v7+ and all updates etc. done.

sudo rasps-update works without complains but after the reboot, the kernel is still the same.

thanks for your effort!

monouser7dig, 2017-03-14:
sometimes the output does vary though 

*** Updating firmware
 *** Updating kernel modules
 *** depmod 4.9.14-v7+
depmod: ERROR: failed to load symbols from //lib/modules/4.9.14-v7+/kernel/drivers/net/wireless/ath/ath9k/ath9k_htc.ko: Exec format error
depmod: ERROR: failed to load symbols from //lib/modules/4.9.14-v7+/kernel/drivers/net/wireless/ath/ath6kl/ath6kl_core.ko: Exec format error
depmod: ERROR: failed to load symbols from //lib/modules/4.9.14-v7+/kernel/drivers/net/wireless/ath/ath6kl/ath6kl_usb.ko: Exec format error
depmod: ERROR: failed to load symbols from //lib/modules/4.9.14-v7+/kernel/drivers/net/wireless/ath/ar5523/ar5523.ko: Invalid argument
depmod: ERROR: failed to load symbols from //lib/modules/4.9.14-v7+/kernel/drivers/net/wireless/ath/ath.ko: Exec format error
depmod: ERROR: failed to load symbols from //lib/modules/4.9.14-v7+/kernel/drivers/net/usb/rtl8150.ko: Invalid argument
depmod: ERROR: failed to load symbols from //lib/modules/4.9.14-v7+/kernel/drivers/net/usb/r8152.ko: Exec format error
depmod: ERROR: failed to load symbols from //lib/modules/4.9.14-v7+/kernel/drivers/net/usb/sr9700.ko: Invalid argument
depmod: ERROR: failed to load symbols from //lib/modules/4.9.14-v7+/kernel/drivers/net/usb/sr9800.ko: Exec format error
depmod: ERROR: failed to load symbols from //lib/modules/4.9.14-v7+/kernel/drivers/net/usb/smsc75xx.ko: Invalid argument
depmod: ../libkmod/libkmod-elf.c:207: elf_get_mem: Assertion `offset < elf->size' failed.
/usr/bin/rpi-update: line 78: 16433 Aborted                 depmod -b "${ROOT_PATH}" -a "${BASEDIR}"

/////And after a second attempt 
*** Raspberry Pi firmware updater by Hexxeh, enhanced by AndrewS and Dom
 *** Performing self-update
/usr/bin/rpi-update: line 65: //root/.updateScript.sh: Input/output error

//////After reboot
 *** Updating firmware
 *** Updating kernel modules
 *** depmod 4.9.14-v7+
 *** depmod 4.9.14+
 *** Updating VideoCore libraries
 *** Using HardFP libraries
 *** Updating SDK
 *** Running ldconfig
 *** Storing current firmware revision
 *** Deleting downloaded files
 *** Syncing changes to disk
 *** If no errors appeared, your firmware was successfully updated to b30461d2621e71eb9b8845b38738af177b9181ed
 *** A reboot is needed to activate the new firmware

// still no updated firmware after reboot though

popcornmix, 2017-03-15:
Can you report output of
```
vcgencmd version
uname -a
mount
```

monouser7dig, 2017-03-15:
actually it seems like the microSD card had an issue that would let devices read the card but not write to it, that is also how it behaved as I connected it to a PC and tried read/ write 

the new setup (new sd) now works perfectly fine 

thank you!

closed 2017-03-15 by popcornmix