Hexxeh/rpi-update #242: Can't properly install rpi-update

AndreiAdrian, 2017-05-06:
I have run the command:
`
sudo apt-get install rpi-update
Reading package lists... Done
Building dependency tree
Reading state information... Done
rpi-update is already the newest version.
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
`

And after that 
sudo rpi-update, but i keep getting this:
`
/usr/bin/rpi-update: 1: /usr/bin/rpi-update: cannot open !DOCTYPE: No such file
/usr/bin/rpi-update: 1: /usr/bin/rpi-update: html: not found
/usr/bin/rpi-update: 2: /usr/bin/rpi-update: cannot open html: No such file
/usr/bin/rpi-update: 3: /usr/bin/rpi-update: cannot open head: No such file
/usr/bin/rpi-update: 4: /usr/bin/rpi-update: cannot open meta: No such file
/usr/bin/rpi-update: 5: /usr/bin/rpi-update: cannot open meta: No such file
/usr/bin/rpi-update: 5: /usr/bin/rpi-update: http-equiv=Pragma: not found
/usr/bin/rpi-update: 6: /usr/bin/rpi-update: cannot open title: No such file
/usr/bin/rpi-update: 7: /usr/bin/rpi-update: cannot open link: No such file
/usr/bin/rpi-update: 8: /usr/bin/rpi-update: cannot open !--: No such file
/usr/bin/rpi-update: 8: /usr/bin/rpi-update: Fav: not found
/usr/bin/rpi-update: 9: /usr/bin/rpi-update: cannot open link: No such file
/usr/bin/rpi-update: 10: /usr/bin/rpi-update: Syntax error: redirection unexpected
`
Could somebody please help me?

popcornmix, 2017-05-06:
Try:
```
sudo wget https://raw.githubusercontent.com/Hexxeh/rpi-update/master/rpi-update -O /usr/bin/rpi-update
```

Ruffio, 2017-05-16:
@AndreiAdrian has this issue been resolved?

popcornmix, 2017-05-30:
Closing due to lack of response.
Please re-open if there is still an issue.

closed 2017-05-30 by popcornmix