Hexxeh/rpi-update #243: Upgrade to specific version using short hash

johnwallace123, 2017-05-19:
For ease of use, it would be convenient if rpi-upgrade could use the short hash to upgrade to a specific revision.  As an example, see result below of checking for latest update (always using "next" branch):

Just checking normally:
```
$ sudo BRANCH=next JUST_CHECK=1 rpi-update
 *** Raspberry Pi firmware updater by Hexxeh, enhanced by AndrewS and Dom
 *** Performing self-update
 *** Relaunching after update
 *** Raspberry Pi firmware updater by Hexxeh, enhanced by AndrewS and Dom
 *** Your firmware is already up to date
```

Checking using long SHA1 hash:
```
$ sudo BRANCH=next JUST_CHECK=1 rpi-update f4cb84915874f32d88e2bfe793620d7afe25c755
 *** Raspberry Pi firmware updater by Hexxeh, enhanced by AndrewS and Dom
 *** Performing self-update
 *** Relaunching after update
 *** Raspberry Pi firmware updater by Hexxeh, enhanced by AndrewS and Dom
 *** Your firmware is already up to date
```

Checking using short hash
```
$ sudo BRANCH=next JUST_CHECK=1 rpi-update f4cb849
 *** Raspberry Pi firmware updater by Hexxeh, enhanced by AndrewS and Dom
 *** Performing self-update
 *** Relaunching after update
 *** Raspberry Pi firmware updater by Hexxeh, enhanced by AndrewS and Dom
 *** Firmware update required. New commits available:
```

popcornmix, 2017-05-19:
Can you test [this version](https://gist.githubusercontent.com/popcornmix/1068a021003a4aee1de0786f33c60bf2/raw/2c07a0edaa406db19d5c886ea0f9f93923e0b66e/rpi-update) and see if it behaves as you desire?
This will update your rpi-update:
```
sudo wget https://gist.githubusercontent.com/popcornmix/1068a021003a4aee1de0786f33c60bf2/raw/2c07a0edaa406db19d5c886ea0f9f93923e0b66e/rpi-update -O /usr/bin/rpi-update
```
Run with ``UPDATE_SELF=0`` to avoid replacing the test version.

johnwallace123, 2017-05-19:
That worked wonderfully!  This worked exactly as expected when moving back and forth between the most recent revision and the prior revision, only using what github shows.

The only tiny, minor issue that I found was that checking the difference when moving "backward" correctly identifies that a new version is needed, but does not report any commits.

Example moving from 379c634 to f4cb849 (HEAD-1 to HEAD of next):
```
$ sudo UPDATE_SELF=0 JUST_CHECK=1 rpi-update f4cb849
 *** Raspberry Pi firmware updater by Hexxeh, enhanced by AndrewS and Dom
 *** Firmware update required. New commits available:
======================================================
Commit: kernel: BCM2835-V4L2: Ensure H264 header bytes get a sensible timestamp\nkernel:  BCM2835-V4L2: Correctly denote key frames in encoded data\nkernel:  bcm2835-gpio-exp: Driver for GPIO expander via mailbox service\nSee: https://github.com/raspberrypi/linux/pull/1852
Commit: kernel: Add overlay for ads1115 ADCs\nSee: https://github.com/raspberrypi/linux/pull/1864
Commit: kernel: Add support for Fe-Pi audio sound card\nSee: https://github.com/raspberrypi/linux/pull/1867
Commit: kernel: amba_pl011: Round input clock up\nkernel: clk-bcm2835: Correct the prediv logic
Commit: firmware: Redo CEC code cleanup 3: Removed CEC topology computation\nfirmware: Redo CEC code cleanup 4: Removed unused functions\nfirmware: Redo CEC code cleanup 5: Removed Rx processing\nfirmware: Redo CEC code cleanup 6: Logging changed to VCOS\nfirmware: Redo CEC code cleanup 7: Removed hdmi_state_machine_clock_changed\nfirmware: Redo CEC code cleanup 8: fixed hdmi state machine clk\nfirmware: Redo CEC code cleanup 10: misc cosmetic changes\nfirmware: Fixup CEC code cleanup 8: fixed hdmi state machine clk\nfirmware: hdmi: Add way of forcing logging for hdmi and cec from boot\nfirmware: Redo CEC code cleanup 11: cec_release_logical_addr
Commit: firmware: arm_loader: Respect smsc95xx.macaddr parameter\nSee: https://github.com/raspberrypi/linux/issues/1865
Commit: firmware: vec: Fix progressive scan composite mode\nSee: https://github.com/raspberrypi/firmware/issues/683
Commit: firmware: vec: PAL_M mode is 525/60\nSee: https://github.com/raspberrypi/firmware/issues/756
Commit: firmware: IL ISP: Support Bayer\nfirmware: IL ISP: Fix error in stride calcs for YUYV formats\nfirmware: video_render: buffer size fixup for ARGB888
Commit: firmware: video_render: Support per-pixel alpha on RGBA input\nSee: https://github.com/waveform80/picamera/issues/199
```

And going in reverse (f4cb849 to 379c634): 
```
$ sudo UPDATE_SELF=0 JUST_CHECK=1 rpi-update 379c634
 *** Raspberry Pi firmware updater by Hexxeh, enhanced by AndrewS and Dom
 *** Firmware update required. New commits available:
```

Of course, this only covers the very rare edge case of checking to see if you need to downgrade to a specific version, so I don't see this as an actual issue.

Thank you so much for the quick response and resolution!  This saves me from needing to figure it out and submitting a PR.

popcornmix, 2017-05-30:
The list of commits comes from:
https://api.github.com/repos/Hexxeh/rpi-firmware/compare/f4cb849...379c634
https://api.github.com/repos/Hexxeh/rpi-firmware/compare/379c634...f4cb849

This api requires a linear path from old commit to new commit.
Obviously you can break that by downgrading or swapping branches.
However the common case of downgrading from the same branch can be detected and we can swap the new/old commits over. I have added this.

I've also pushed the change you tested previously. Please update and confirm if all is okay.

notro, 2017-05-31:
This seems to have broken the ability to update from a local repo:
```
sudo UPDATE_SELF=0 SKIP_BACKUP=1 SKIP_REPODELETE=0 SKIP_DOWNLOAD=1 FW_REPOLOCAL=rpi-build-archive rpi-update '2017-05-31 21:07:43 +0200' 1>&2
 *** Raspberry Pi firmware updater by Hexxeh, enhanced by AndrewS and Dom
 *** Invalid hash given
```

My build script uses rpi-update (over ssh) to push out kernels for testing locally.

popcornmix, 2017-05-31:
@notro I don't think it's related to using a local repo, but it's the use of '2017-05-31 21:07:43 +0200' as a hash.
We look up:
```
https://api.github.com/repos/Hexxeh/rpi-firmware/commits/<hash>
```
which your date isn't valid for. I don't think what you are passing in has ever been supported - it just didn't check it before. 
Can you pass in the hash of the repo you downloaded?
Otherwise we could give an option or magic hash value (``-``) that doesn't get checked.

notro, 2017-06-01:
I don't download a repo, I create a folder locally, push it to the board and use rpi-update to deploy the kernel/firmware for testing. The reason I have used current time as a ref, was to ensure it was always installed.

It works if I delete /boot/.rpi-firmware prior to calling rpi-update _without_ a ref. Thanks.

lurch, 2019-06-08:
I guess this can be closed @johnwallace123

Ruffio, 2020-12-23:
@johnwallace123 can this be closed?

closed 2021-02-19 by johnwallace123