Hexxeh/rpi-update #244: Error: function strtonum never defined

steffanson, 2017-05-30:
Hi!

Since updating from https://github.com/raspberrypi/firmware/commit/73f44c607b48b6c7af89aa3cc034fa3d93d98379 to https://github.com/raspberrypi/firmware/commit/b9dbf8cf1a4609d16304c3f2740e97d5057d9a1a I see the following error on my RaspberryPi 2, running Raspbian 8:
```
awk: line 2: function strtonum never defined
awk: line 2: function strtonum never defined
awk: line 2: function strtonum never defined
(23) Failed writing body
```

when I invoke `sudo rpi-update`. The latest updates are installed via apt-get.

Do you have any idea what causes this error or how I can debug or resolve it? If you need more information, please let me know.

popcornmix, 2017-05-30:
What does ``awk --version`` report for you?

Whobeu, 2017-05-30:
Have the same problem with Wheezy.  "awk --version" says "awk: not an option: --version".  Man page says it is version 1.2.

popcornmix, 2017-05-30:
Can you report output of:
```
which awk
awk
```

Linux-Maverick, 2017-05-30:
Same problem here.
**which awk**
/usr/bin/awk
**awk**
**awk -W version**
mawk 1.3.3 Nov 1996, Copyright (C) Michael D. Brennan

compiled limits:
max NF             32767
sprintf buffer      1020

steffanson, 2017-05-30:
Weird, I haven't uninstalled awk; however it was no longer installed.
```
Package awk is a virtual package provided by:
  original-awk 2012-12-20-2
  mawk 1.3.3-17
  gawk 1:4.1.1+dfsg-1
You should explicitly select one to install.

E: Package 'awk' has no installation candidate
```
I now installed `GNU Awk 4.1.1, API: 1.1 (GNU MPFR 3.1.2-p3, GNU MP 6.0.0)` and `rpi-update` works again.

Steltek, 2017-05-30:
Same problem on Debian Jessie. manpage describes it as 'mawk'. Package manager says 'mawk v1.3.3-17'

popcornmix, 2017-05-30:
Strange. I have:
```
pi@domnfs:~ $ awk --version 
GNU Awk 4.1.1, API: 1.1 (GNU MPFR 3.1.2-p3, GNU MP 6.0.0)
Copyright (C) 1989, 1991-2014 Free Software Foundation.
```
But seems some users have mawk as the default awk.

Anyway can you test this version of rpi-update which avoid strtonum
```sudo wget -O "/usr/bin/rpi-update https://drive.google.com/uc?id=0B-6zmEDJwxZEYUNYbXZUa01aNXM&export=download"

Whobeu, 2017-05-30:
Same as Linux-Maverick noted above.  /usr/bin/awk and "1.3.3 Nov 1996".

Linux-Maverick, 2017-05-30:
Hmm. I get the same results from the new file. Grep still reveals references to strtonum in the script (in GITHUB_API_LIMITED=), but it is definitely the one that I just retrieved from your URL above.

popcornmix, 2017-05-30:
Ah download again and run with:
```
sudo UDATE_SELF=0 rpi-update
```
otherwise it will overwrite your changed version.

Linux-Maverick, 2017-05-30:
Ran it with the same results, then saw the typo and re-ran:
**UPDATE_SELF=0 rpi-update**
Seems to be working as we speak!
Thank you for getting on this so quickly, popcornmix!!!

popcornmix, 2017-05-30:
Thanks for testing. I've pushed, so now ``rpi-update`` should be fixed with mawk.

Linux-Maverick, 2017-05-30:
As a final note, the arguments for wget above didn't work exactly for me. I ended up just doing a wget on your file URL to my local home directory, changed the file name, saved copies of both the original file and the new one, then moved the new one to /usr/bin. The update ran flawlessly, and the pi rebooted and came up with the new kernel just fine.
Thanks again!

popcornmix, 2017-05-30:
> As a final note, the arguments for wget above didn't work exactly for me.

Ah yes, I've edited the post to fix the quoting.

Steltek, 2017-05-30:
Can confirm, after manually pulling the new update, it's back to working order.

popcornmix, 2017-05-30:
@steffanson are you up and running? Okay to close?

steffanson, 2017-05-30:
@popcornmix  Yes, I could solve it for myself by installing awk again. Okay to close!

closed 2017-05-30 by steffanson

Atomixcl, 2018-06-07:
Hello: getting this error: "awk: line 2: function strftime never defined".
Package awk is a virtual package provided by:
  original-awk:i386 2012-12-20-5
  original-awk 2012-12-20-5
  mawk:i386 1.3.3-17ubuntu2
  gawk:i386 1:4.1.3+dfsg-0.1
  mawk 1.3.3-17ubuntu2
  gawk 1:4.1.3+dfsg-0.1
You should explicitly select one to install.

sudo apt install gawk, then:

date +"%s" | awk '{ print strftime("%d/%m/%Y %H:%M:%S",$1)}'
07/06/2018 12:16:19

Ubuntu 16.04, up to date.

popcornmix, 2018-06-08:
Are you running rpi-update on Ubuntu?
Are you saying you still have the issue after installing gawk?
What does ``awk --version`` report?
Can you show complete output of running rpi-update?

Atomixcl, 2018-06-08:
Sorry, I wasn't explicit enough. I got the problem for strftime, and I solved it installing gawk (following the same instruction above). Then the command estar working:
date +"%s" | awk '{ print strftime("%d/%m/%Y %H:%M:%S",$1)}'
07/06/2018 12:16:19

Done at Ubuntu 16.04, with packages Up to date.

Sorry, popcornmix for the inconvenience. Thanks for the previous post, save me a lot of time.