Hexxeh/rpi-update #245: (23) Failed writing body

MrEngman, 2017-06-04:
I've started seeing this apparent error when using `rpi-update commit-id` to revert to an older version of the raspbian kernel.

```
 *** Raspberry Pi firmware updater by Hexxeh, enhanced by AndrewS and Dom
 *** Performing self-update
 *** Relaunching after update
 *** Raspberry Pi firmware updater by Hexxeh, enhanced by AndrewS and Dom
(23) Failed writing body
WARNING: This update bumps to rpi-4.1.y linux tree
Be aware there could be compatibility issues with some drivers
```
A bit more detail here []https://www.raspberrypi.org/forums/viewtopic.php?p=1171152#p1171152

popcornmix, 2017-06-04:
I didn't see this issue. Do you have eny environment variables set that rpi-updates uses?
Do you get this every time?
What was in ``/boot/.firmware_revision`` Do you still get the issue if that file is removed?
```
pi@domnfs:~ $ sudo rpi-update 9037cc8c8af19134d821fc06e27b10be2ad98a61
 *** Raspberry Pi firmware updater by Hexxeh, enhanced by AndrewS and Dom
 *** Performing self-update
 *** Relaunching after update
 *** Raspberry Pi firmware updater by Hexxeh, enhanced by AndrewS and Dom
WARNING: This update bumps to rpi-4.1.y linux tree
Be aware there could be compatibility issues with some drivers
Discussion here:
https://www.raspberrypi.org/forums/viewtopic.php?f=29&t=113753
##############################################################
Would you like to proceed? (y/N)

 *** Downloading specific firmware revision (this will take a few minutes)
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100   168    0   168    0     0    269      0 --:--:-- --:--:-- --:--:--   270
100 47.0M    0 47.0M    0     0   981k      0 --:--:--  0:00:49 --:--:--  566k
 *** Updating firmware
 *** Updating kernel modules
 *** depmod 4.1.4-v7+
 *** depmod 4.1.4+
 *** Updating VideoCore libraries
 *** Using HardFP libraries
 *** Updating SDK
 *** Running ldconfig
 *** Storing current firmware revision
 *** Deleting downloaded files
 *** Syncing changes to disk
 *** If no errors appeared, your firmware was successfully updated to 9037cc8c8af19134d821fc06e27b10be2ad98a61
 *** A reboot is needed to activate the new firmware
```

MrEngman, 2017-06-04:
I was running a Pi B+ with kernel 4.9.30+ #1001 and then ran `sudo rpi-update 7c75bc049eef5c351a23919f65060b0e1a4a2181` to revert to kernel 4.0.5 #797 to check a wifi driver when I first saw this issue.

I then ran `sudo rpi-update 2344e11340a8a26dbb68f2356845b13e68655233` and saw it again.

Then ran `sudo rpi-update 9037cc8c8af19134d821fc06e27b10be2ad98a61` and saw it again.

However, just ran `sudo rpi-update 771a503cfc2a1130e2df2a4ddfc45ffa0f592b3f` and the issue disappeared.

Weird.

popcornmix, 2017-06-04:
The error message comes from curl and means it couldn't write to the output device.
The simplest explanation is sdcard full, but I suspect it is something else.

It can come from piping output of curl to another process that exits before curl has finished. I only see one case where that may happen, but it isn't going wrong for me.
Can you try running this a number of times and report if it gives the error?
```
curl -Ls https://api.github.com/repos/Hexxeh/rpi-firmware/commits/master | awk '{ if ($1 == "\"sha\":") { print substr($2, 2, 40); exit 0 } }'
```

MrEngman, 2017-06-04:
Try a dozen or so times and it just returns the latest hexxeh rpi-firmware commit-id `684be4bc8cc343f60fdc3240c6d55d41d0a5b56c`

My SD card is 64Gb and only about 9% used so plenty of room.

Just tried the previous commands that gave the original error `sudo rpi-update 6d25078b7cb361052ae280d552561d7250d33f5b`

and got 
```
 *** Raspberry Pi firmware updater by Hexxeh, enhanced by AndrewS and Dom
 *** Performing self-update
 *** Relaunching after update
 *** Raspberry Pi firmware updater by Hexxeh, enhanced by AndrewS and Dom
WARNING: This update bumps to rpi-4.0.y linux tree
Be aware there could be compatibility issues with some drivers
Discussion here:
https://www.raspberrypi.org/forums/viewtopic.php?f=29&t=113753
##############################################################
Would you like to proceed? (y/N)
```
Then tried 
```
pi@Pi-64GB-1:~ $ sudo rpi-update 7c75bc049eef5c351a23919f65060b0e1a4a2181
 *** Raspberry Pi firmware updater by Hexxeh, enhanced by AndrewS and Dom
 *** Performing self-update
 *** Relaunching after update
 *** Raspberry Pi firmware updater by Hexxeh, enhanced by AndrewS and Dom
(23) Failed writing body
WARNING: This update bumps to rpi-4.0.y linux tree
Be aware there could be compatibility issues with some drivers
Discussion here:
https://www.raspberrypi.org/forums/viewtopic.php?f=29&t=113753
##############################################################
Would you like to proceed? (y/N)
```
and it re-appeared. Rebooted the Pi and tried again and it appeared again.

Rather weird, however it doesn't appear to cause problems updating the firmware.  

Just tried `sudo rpi-update 2f2b1e7e69fdd33b4ca2394b40858b4aabff2494` and the issue disappeared.
```
pi@Pi-64GB-1:~ $ sudo rpi-update 2f2b1e7e69fdd33b4ca2394b40858b4aabff2494
 *** Raspberry Pi firmware updater by Hexxeh, enhanced by AndrewS and Dom
 *** Performing self-update
 *** Relaunching after update
 *** Raspberry Pi firmware updater by Hexxeh, enhanced by AndrewS and Dom
WARNING: This update bumps to rpi-4.4.y linux tree
Be aware there could be compatibility issues with some drivers
Discussion here:
https://www.raspberrypi.org/forums/viewtopic.php?f=29&t=144087
##############################################################
Would you like to proceed? (y/N)
```
and then
```
pi@Pi-64GB-1:~ $ sudo rpi-update 7c75bc049eef5c351a23919f65060b0e1a4a2181
 *** Raspberry Pi firmware updater by Hexxeh, enhanced by AndrewS and Dom
 *** Performing self-update
 *** Relaunching after update
 *** Raspberry Pi firmware updater by Hexxeh, enhanced by AndrewS and Dom
(23) Failed writing body
WARNING: This update bumps to rpi-4.0.y linux tree
Be aware there could be compatibility issues with some drivers
Discussion here:
https://www.raspberrypi.org/forums/viewtopic.php?f=29&t=113753
##############################################################
Would you like to proceed? (y/N)
```

popcornmix, 2017-06-04:
I'm just guessing but I've pushed a change to rpi-update. Any different?

MrEngman, 2017-06-04:
Looks like you've fixed the issue. Tried a number of different commits that gave the error before and no sign of it now. Thanks.

popcornmix, 2017-06-04:
I'll close for now. Feel free to re-open if the problems reoccurs.

closed 2017-06-04 by popcornmix