Hexxeh/rpi-update #248: Update to 4.34 with Raspberry Pi Zeros

rubensutter, 2017-06-28:
Updating to 4.34 resulted in strange behaviour yesterday. In my case only Raspberry Pi Zeros no longer boot kernel modules rebooting after rpi-update. One of the Rasperry Pi Zeros showed the wrong Kernel Version after reboot using uname -a. Network is unavailable due to missing USB Wifi/Network functionality(kernel modules). Other Raspberries (not of Model Zero) show the new kernel version and still work as expected. All Zeros are running Debian 9 Stretch.

popcornmix, 2017-06-28:
It sounds like the /boot/kernel.img has failed to update.
Possibly due to powering off device without shutting down or some other form of sdcard corruption.
Can you compare your version (e.g. use md5sum) with the expected version (from https://github.com/Hexxeh/rpi-firmware/blob/master/kernel.img)?
You may want to run scandisk (windows) or fsck (linux) on the sdcard to check it is okay.
After that try copying a newly downloaded kernel.img onto sdcard.

rubensutter, 2017-06-28:
Thanks, I just checked the md5sums:

SD-Card File: 4b3b1f23491a2ad8c168242055afd772
expected:      8d62def5a3a8afed44ee533c52fd34c6

In this case I can also find some unusual data in the boot directory,
!ERNEL~1.TMP, kernel.img.dpkg-diver.tmp,
kernel.img.FileSlack and some .dpkg-divert.tmp and .FileSlack files.

I'll try to fix the files in the boot folder and see if this one boots up again.

rubensutter, 2017-06-28:
I cleaned up the boot directory on microSD, 
then I copied the master files directly to microSD.
This Zero is now up and running smoothly again. 
No error messages while booting.
All USB Devices and Network up and running.

popcornmix, 2017-06-29:
To be sure you may want to try rpi-update again. Run:
```
sudo /boot/.firmware_revision
```
before running rpi-update to force it to update (even if it thinks you are up to date).

lurch, 2019-06-08:
@popcornmix I guess there was an `rm` missing from your command above? :wink:

Ruffio, 2020-12-23:
@rubensutter is this stil relevant or can the issue be closed?

lurch, 2020-12-23:
The original issue was opened over 3 years ago, so it can presumably be closed now.

rubensutter, 2020-12-24:
@Ruffio this issue only occured once quite some time ago and can be closed.

closed 2020-12-24 by rubensutter