Hexxeh/rpi-update #250: RPi 2B Update stops when downloading!

Xie-Tian, 2017-08-09:
Hi,
When I run rpi-update it doesn't seem to be finishing the downloads for the updates it is doing. 
It ends with following:
```
gzip: stdin: unexpected end of file
tar: Unexpected EOF in archive
tar: Unexpected EOF in archive
tar: Error is not recoverable: exiting now
```
Also, my ssh to other IP address `ssh pi@IPaddress` will also disconnect in several seconds after successfully connecting.
Is this a local network issue? But I tried ping to other ip address `ping IP address,` it can continue working for at least 10 minutes.
Does anyone have the idea?
Thanks!

popcornmix, 2017-08-10:
This sounds like a local network issue. Nothing to do with rpi-update.

You can test this just by trying to download the firmware directly from pi:
```
wget  https://github.com/Hexxeh/rpi-firmware/tarball/master
```

Xie-Tian, 2017-08-10:
I tried for 4 hours, but it just continued loading and ended with "Read error at byte xxxxxx Resource temporarily unavailable, try again. Retrying". And begin with another loading....

popcornmix, 2017-08-12:
Are you connecting Pi through wired ethernet or wifi?

Xie-Tian, 2017-09-01:
Thanks! No problem now because I don't need to update my pi. Thanks a lot!

closed 2017-09-01 by Xie-Tian