Hexxeh/rpi-update #251: Weird USB/network issue after rpi-update

geezuh, 2017-08-20:
I just ran rpi-update because my pi didn't seem to have bcm_host.h which I needed to build mpv.

It froze in the middle of the update. This is the last thing it said:
*** Using HardFP libraries  

After noticing I couldn't ping the device or get it to respond in any way, I pulled the power and did a hard reboot. It came back up, and seemed to be normal, but it is unresponsive to keyboard/mouse input and is no longer connected to wifi. I tried both my bluetooth keyboard which was already paired, and a usb keyboard as well.

Is there a way to revert this? Perhaps I could put a script somewhere on the SD card to restore the backups rpi-update made? I'd prefer not to have to get back to where I am from scratch if possible

I'm running a Pi3 from the Retropie Distro (based on raspbian) whatever was current 6 months or so ago.

lurch, 2019-06-08:
If you have a working SD card and a USB SD-reader, you can boot off the working card, put the 'broken' card in the SD-reader, mount the partitions from the 'broken' card (e.g. to `/mnt/root` and `/mnt/boot`) and then use the "offline" update options https://github.com/Hexxeh/rpi-update#root_path-and-boot_path to try again updating the firmware on the 'broken' card.

Ruffio, 2020-12-25:
@geezuh is this issue still relevant or can it be closed?

geezuh, 2020-12-25:
I gave up on the project I was working on. I'm not sure if this is still an
issue or not.

Ruffio, 2020-12-25:
@geezuh ok, then please close this issue. Thanks :-)

closed 2020-12-26 by geezuh